volumio/Volumio2 #1149: No ssh connection possible after upgrading to v12

masterkyle, 2017-04-19:


gwlevans, 2017-04-19:
Try logging in as user volumio, password volumio, then use sudo for root commands. I think SSH as root is disabled by default.

smalahove, 2017-04-20:
Same as for masterkyle here. Solution found here, i.e. deleting system --> userdata
https://github.com/volumio/Volumio2/issues/1045

masterkyle, 2017-04-21:
> Try logging in as user volumio, password volumio, then use sudo for root commands
did you read my description?
@smalahove 
Thanks for your hint, will try it

smalahove, 2017-04-21:
@masterkyle - I was unable to access the box via ssh after upgrade. I normally use ssh-keys, so login is automatic, but after upgrade I could not log in. I tried to force password both for root  and volumio users, same result. Simple network commands such as ping segfaulted, so I expect that is what happened to the ssh server: it, or some related network component, segfaults when you try to log in. I was successfully able to ssh to localhost on the box, but once I tried from the outside, it did not work.

That seems to be exactly the same problem as described in the post I linked to, and the solution did work fine for me. The drawback of course being you have to redo some of your settings.

Hope you get it working again!

masterkyle, 2017-04-21:
I tried deleting user data, now the system doesn't boot anymore. It stucks at the login screen after booting.
I think I've to reflash the sd.

balbuze, 2017-04-21:
You said it doesn't boot.., and you said it stuck on login.... It can't be both... And you can't use volumio login into it that way. It is just if you want to tweak or install other software. The normal use is to connect through a web browser from a other device on the same network. Enter the IP of your volumio device in the url and that's it!

closed 2017-06-02 by volumio