volumio/Volumio2 #1154: wireless config broken after update to 2.141

nilsbecker, 2017-04-29:
i just updated and cannot get a wireless connection anymore. (rpi3) ethernet works. 
when i use the web ui to activate wireless, i get the option to switch it on. 
after switching on and saving, no wireless networks are listed
after navigating away and back to the network settings page, the wireless is listed as off again.

i imagine the wireless config got overwritten/messed up in the update. what can i do to reset it?

xipmix, 2017-05-02:
Are you able to send in some logs of what happens when you try switching on the wireless?
 * reboot
 * try to activate wireless in web ui
 * go to /dev page, press the 'send logs' button near the bottom

macmpi, 2017-05-02:
@nilsbecker : can you try with a freshly imaged SD (preferred), or clear all user data with Web UI, and check if problem persists?

nilsbecker, 2017-05-02:
hi, i am having problems with my isp right now. i will get back to you when i can.

leafy7382, 2017-05-03:
I am running into similar issue.

1. Plug into wired, can configure wireless now and getting 72Mbps
2. Unplug wired without rebooting, wireless still works, retains the same IP.
3. Now, reboot without wired plugged in, wireless does not reconnect properly
4. Plugging the wired connection back in does not yield proper IP either, so there is no way to get a /dev log sorry

macmpi, 2017-05-03:
@leafy7382 which device are you using, what are the network interfaces (built-in, dongles,...)?
Can you plug a monitor & keyboard to the devices so we can check things when there is no connection?

volumio, 2017-05-03:
@macmpi I confirm that PI3's internal wireless is broken. Wireless works only if an external dongle is used. Could you investigate?

macmpi, 2017-05-03:
Happy to help, but I do not have a Pi3: need to find someone who can share connection to one and try remotely...

Is it due to udev rules: does it work if `/etc/udev/rules.d/99-Volumio-net.rules` is removed (renamed without `.rules`)?

macmpi, 2017-05-03:
Does the issue appear with newly re-imaged SD, or only with OTA update?

volumio, 2017-05-03:
Newly flashed img. This one:
http://updates.volumio.org/pi/volumio/2.156/volumio-2.156-2017-05-03-pi.img.zip

Renaming 99-Volumio-net.rules does not help 

What device do you have for testing? I assumed you had a PI3 since you told it was tested...

macmpi, 2017-05-03:
Yes tested on a Pi3 remotely with help of @chsims1, and another one.
As it does not seem to come from udev rule, maybe some recent firmware file changes could be a cause.
Can one try using firmware files (from `/boot`) from earlier working version (2.129 for instance)?

volumio, 2017-05-03:
Sure, specifically which files am I looking for?

macmpi, 2017-05-03:
*.elf *.dat *.bin

macmpi, 2017-05-03:
Does `ifconfig` display coherent interface names?
Can one share a log saved onto SD under console when no connections is available?

volumio, 2017-05-03:
Ifconfig shows no wlan0 . 
However something strange is going on here... Flashed previous image (2.111) and tried on 2 different pis to exclude hardware issues.... Same result: both modules loaded for bcrm, no wlan0.
Now, trying with a third pi which shows wlan0 but creates no hotspot...
I'll keep you posted...

Are there reports of broken wifi on PI3s?

volumio, 2017-05-03:
More news on this:
- With my third PI3 now wifi works
- Still no hotspot is started on first boot, seems an issue with /bin/hotspot.sh
-Trying with an older version of /bin/hotspot.sh

volumio, 2017-05-03:
Ok, after restoring old /bin/hotspot.sh this is what I get:

May 03 22:37:55 volumio kernel: brcmfmac: brcmf_do_escan: error (-52)
May 03 22:37:55 volumio kernel: brcmfmac: brcmf_cfg80211_scan: scan error (-52)
May 03 22:37:55 volumio volumio[1057]: An error occurred while scanning: Error: Command failed: sudo /sbin/iwlist wlan0 scan
May 03 22:37:55 volumio volumio[1057]: wlan0     Interface doesn't support scanning : Invalid exchange
May 03 22:37:55 volumio volumio[1057]: Cannot use regular scanning, forcing with ap-force
May 03 22:37:55 volumio sudo[1471]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/iw dev wlan0 scan ap-force
May 03 22:37:55 volumio sudo[1471]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:37:57 volumio sudo[1471]: pam_unix(sudo:session): session closed for user root
May 03 22:37:57 volumio kernel: brcmfmac: brcmf_do_escan: error (-52)
May 03 22:37:57 volumio kernel: brcmfmac: brcmf_cfg80211_scan: scan error (-52)
May 03 22:37:57 volumio volumio[1057]: command failed: Invalid exchange (-52)
May 03 22:37:57 volumio volumio[1057]: Cannot use fallback scanning method: Error: Command failed: /usr/bin/sudo /sbin/iw dev wlan0 scan ap-force
May 03 22:37:57 volumio volumio[1057]: command failed: Invalid exchange (-52)
May 03 22:38:08 volumio volumio[1057]: info: CALLMETHOD: system_controller network saveHotspotSettings [object Object]
May 03 22:38:08 volumio volumio[1057]: info: CoreCommandRouter::executeOnPlugin: network , saveHotspotSettings
May 03 22:38:08 volumio sudo[1480]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/hostapd/hostapd-edimax.conf
May 03 22:38:08 volumio sudo[1480]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:08 volumio sudo[1482]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/hostapd/hostapd.conf
May 03 22:38:08 volumio sudo[1480]: pam_unix(sudo:session): session closed for user root
May 03 22:38:08 volumio sudo[1482]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:08 volumio sudo[1482]: pam_unix(sudo:session): session closed for user root
May 03 22:38:08 volumio volumio[1057]: info: Permissions for /etc/hostapd/hostapd-edimax.conf
May 03 22:38:08 volumio volumio[1057]: info: Permissions for /etc/hostapd/hostapd.conf
May 03 22:38:08 volumio sudo[1498]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl restart wireless.service
May 03 22:38:08 volumio sudo[1498]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:08 volumio systemd[1]: Starting Wireless Services...
May 03 22:38:08 volumio sudo[1504]: root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/iw dev wlan0 set power_save off
May 03 22:38:08 volumio sudo[1504]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:08 volumio sudo[1504]: pam_unix(sudo:session): session closed for user root
May 03 22:38:08 volumio wireless.js[1513]: start
May 03 22:38:08 volumio wireless.js[1513]: Cleaning previous...
May 03 22:38:08 volumio systemd[1]: Stopping dhcpd.service...
May 03 22:38:08 volumio systemd[1]: Stopping Host and Network Name Lookups.
May 03 22:38:08 volumio systemd[1]: Stopped target Host and Network Name Lookups.
May 03 22:38:08 volumio systemd[1]: Stopping dnsmasq.service...
May 03 22:38:08 volumio dnsmasq[1052]: exiting on receipt of SIGTERM
May 03 22:38:08 volumio systemd[1]: Stopped dnsmasq.service.
May 03 22:38:08 volumio systemd[1]: Stopped dhcpd.service.
May 03 22:38:08 volumio systemd[1]: Stopping hotspot.service...
May 03 22:38:13 volumio hotspot[1524]: Killing Hostapd
May 03 22:38:13 volumio sudo[1526]: root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/killall hostapd
May 03 22:38:13 volumio sudo[1526]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:13 volumio sudo[1526]: pam_unix(sudo:session): session closed for user root
May 03 22:38:13 volumio hotspot[1524]: Killing Dhcpd
May 03 22:38:15 volumio kernel: brcmfmac: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -52
May 03 22:38:18 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: SET SSID error (-52)
May 03 22:38:20 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: BRCMF_C_DOWN error -52
May 03 22:38:22 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: setting AP mode failed -52
May 03 22:38:24 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: setting INFRA mode failed -52
May 03 22:38:26 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: restoring REGULATORY setting failed -52
May 03 22:38:28 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: BRCMF_C_UP error -52
May 03 22:38:30 volumio kernel: brcmfmac: brcmf_set_mpc: fail to set mpc
May 03 22:38:32 volumio dhcpcd[664]: wlan0: carrier lost
May 03 22:38:32 volumio netplugd[1055]: wlan0: state ACTIVE flags 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000 -> 0x00001003 UP,BROADCAST,MULTICAST
May 03 22:38:32 volumio netplugd[1536]: /etc/netplug/netplug wlan0 out -> pid 1536
May 03 22:38:32 volumio sudo[1534]: root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/killall dhcpd
May 03 22:38:32 volumio systemd[1]: Stopping Wireless Services...
May 03 22:38:32 volumio sudo[1534]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 03 22:38:32 volumio systemd[1]: Starting Wireless Services...
May 03 22:38:32 volumio sudo[1534]: pam_unix(sudo:session): session closed for user root
[... cut ...]
May 03 22:40:09 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: SET SSID error (-52)
May 03 22:40:11 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: BRCMF_C_DOWN error -52
May 03 22:40:12 volumio volumio[1057]: info: CoreCommandRouter::executeOnPlugin: upnp , onRestart
May 03 22:40:13 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: setting AP mode failed -52
May 03 22:40:15 volumio kernel: brcmfmac: brcmf_cfg80211_stop_ap: setting INFRA mode failed -52

Seems something terribly wrong, but can't understand what

macmpi, 2017-05-03:
Can you try with 2.129 which was the last known-good release?
Then can you put 2.141 squash onto that same SD and wipe all data? This would use exact same 2.129 firmware & overlays with 2.141 (I assume those 2 versions have unchanged initramfs, not sure it's changed in 2.156 image)

volumio, 2017-05-03:
Yep, I'll try that...

volumio, 2017-05-04:
As weird as it might sound, I got 3 fault PIs...
Tried with the 4th one, and it just works... No issue whatsoever.
I think tomorrow we should start a thread on the forum to understand how many people have broken pi's wifi...

nilsbecker, 2017-05-04:
i tried again. with a different non-volumio sd card i can still use the wifi. so on my Pi3, the volumio update did not brick the internal wifi. it's only nonfunctional

macmpi, 2017-05-04:
I'm a bit confused, not sure I understand:
Are the 3 failing with 2.129 too, or is fail only with 2.141 and later?
Is the 4th one working with plain 2.141 / 2.129 ?

This kind of reminds-me of some older articles read about some tiny HW differences across some batches of Pi3 which required updated firmware files (can't find the exact reference of these articles right now).
Current 4.4.9 build does a bit of mix&match of firmware files to accommodate old kernel with more recent devices & related dtbs: maybe we touched some limits with the recently-created images.

Do you have same issue if you build image with the 4.9.25 (and same Volumio2 commit references)?

macmpi, 2017-05-04:
@nilsbecker did you try with a re-imaged Volumio SD (not OTA updated)?
Which Volumio version was working fine on your device before?

nilsbecker, 2017-05-04:
no, haven't had the time to re-flash. the previously working version was 2.129.

macmpi, 2017-05-04:
@volumio it turns-out last week stable branch of https://github.com/Hexxeh/rpi-firmware/ moved to 4.9.24!...
so we probably have big mixed bag of kernel / firmware files in images built since!...

volumio, 2017-05-04:
@macmpi is there a way to download an exact firmware like we do with the kernel? 
aka using the commit hash?

macmpi, 2017-05-04:
Probably, I'll look at it.
In the meantime, to confirm, can you try keeping all `/boot` files from 2.129  and just replace squash with 2.141 and check it works on your devices?

volumio, 2017-05-04:
sure, give me few hours and I'll do that . Now my wireless is working with 2.156

macmpi, 2017-05-04:
> Now my wireless is working with 2.156

doing what?
Mixed stuff will produce unpredictable results anyway: now we see it with wireless, but DAC support is going to be terrible too...

volumio, 2017-05-04:
Internal wireless of PI3 is working. 
I'll do the checks, then we will stick with a firmware version once you find it, what do you think

macmpi, 2017-05-04:
Yes: I'm investigating about firmware revisions.
So you confirm the recent udev rules changes are not suspected anymore?

volumio, 2017-05-04:
Yes, we can rule out the udev rule change. 
Two things that could interfere are: 
- The GPIO PR (we're changing permissions here and there, including SPI and UART, used by wifi)
- The 2 commits on hotspot.sh

macmpi, 2017-05-04:
Can you check with this https://github.com/volumio/Build/pull/191 ?
This should fix the firmware discrepancies (BTW PiZero W users were also complaining, probably same root cause)

nilsbecker, 2017-05-05:
i could test flash or opgrade something if desired...

macmpi, 2017-05-05:
@nilsbecker you may refer to [this](https://volumio.org/forum/trouble-connecting-zero-home-wifi-t6471-10.html#p32330) for an interim fix, until a new version is available.

nilsbecker, 2017-05-05:
thanks @macmpi .

volumio, 2017-05-05:
Tried with newer kernel, nope... wifi still broken... And furthermore, it seems quite a common problem

macmpi, 2017-05-05:
So you still have 1 Pi3 working, and 3 not working?
Can you try this under console after boot:
```
sudo modprobe -r brcmfmac
sudo modprobe brcmfmac
```
does it help?

volumio, 2017-05-05:
Nope.. no wifi if I do ifconfig -a

macmpi, 2017-05-06:
@nilsbecker did it fix your problem?
If so, can you please close this issue?
Thanks.

nilsbecker, 2017-05-07:
ok, i updated to 2.163. i had to do the `modprobe brcmfmac` game above, then i got a working wireless connection back. great!

closed 2017-05-07 by nilsbecker

macmpi, 2017-05-07:
@nilsbecker : very interesting thanks.
So without driver unload/reload your Pi3 can not operate with onboard wifi?
Do you have to do it at each reboot then?

nilsbecker, 2017-05-07:
rebooting now...

macmpi, 2017-05-07:
...and also...how did you "upgrade": reflash or OTA?

nilsbecker, 2017-05-07:
OTA

macmpi, 2017-05-07:
OTA won't change firmware files I'm afraid, so you may still have the original issue.
Can you reflash with 2.163?

nilsbecker, 2017-05-07:
yes, i spoke too soon, after a new reboot it's not working. interestingly on the first boot after OTA it did. hmm. so i'll have to flash.

nilsbecker, 2017-05-07:
is there a way to preserve the volumio home dir and other installed packages?

macmpi, 2017-05-07:
you can save/restore in a hack-ish manner yes (for now): look at [this](https://github.com/volumio/Volumio2/issues/1048#issuecomment-288077276)

nilsbecker, 2017-05-07:
ok, i flashed 2.163 and was able to get a wireless connection working. atm i have problems with the web interface stalling after a while, but this is not related to getting a wireless connection, which i confirm works.