volumio/Volumio2 #1156: REST command 'next' beyound last breaks volumio

heidou7p, 2017-05-01:
Hello,

I was writing a remote control python script using REST API, and bumped into some issues. This one is about 'next'. I will report about another one about 'prev' soon after.

Confirmed on: Raspberry Pi 3B + Volumio 2.141

*** 'Next' beyond last breaks volumio ***

Reproduce procedure:

    From web interface, go 'browse', choose a song and 'Clear and play'. I have chosen from my SMB mounted library. This creates a one-song queue.

    Ssh into volumio, and type:

    wget -O- http://localhost:3000/api/v1/commands/?cmd=next 2>/dev/null

    This prints something like '{"time":1493606231667,"response":"next Success"}', with exit code 0. Music stops and web interface shows 'play' button insted of 'pause', which is unresponsive; pushing this button does nothing. Volumio becomes unable to play music unless you reboot it. As a matter of fact, you can reproduce this without REST. Just push next on web interface and it shows same symptom.

Log from 'sudo journalctl -f':

    https://gist.github.com/heidou7p/717c8d5ba0909e814295effc3f8f4bdb

xipmix, 2017-09-04:
I tried to reproduce this on rpi2+2.246+some recent ui [work](https://github.com/volumio/Volumio2-UI/commit/082435c0e92c81631fdcc85e7fd1518e1e55d10e). I think this issue may be fixed now.
If you have a test case that still breaks the interface, please post it.

First I tested with the web ui.

 * ```systemctl restart volumio```
 * one song in queue, not playing. The 'play' icon is shown
 * press 'next' in web UI
 * nothing obvious happens. The 'play' icon is still shown.
 * log file shows
```
Sep 04 12:53:26 volumio volumio[17137]: info:
Sep 04 12:53:26 volumio volumio[17137]: [1504529606871] ---------------------------- Client requests Volumio next
Sep 04 12:53:26 volumio volumio[17137]: info: CoreCommandRouter::volumioNext
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606878] CoreStateMachine::next
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606882] CoreStateMachine::stop
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606886] CoreStateMachine::setConsumeUpdateService undefined
Sep 04 12:53:26 volumio volumio[17137]: UNSET VOLATILE
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606896] ------------------------------ 25ms
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606897] CoreStateMachine::play index undefined
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606899] CoreStateMachine::setConsumeUpdateService undefined
Sep 04 12:53:26 volumio volumio[17137]: info: [1504529606904] CorePlayQueue::getTrack 1
```

Next test:
 * press 'play' icon.
 * 'pause' icon becomes visible.
 * playback starts
 * log file:
```
Sep 04 12:57:33 volumio volumio[17137]: info:
Sep 04 12:57:33 volumio volumio[17137]: [1504529853635] ---------------------------- Client requests Volumio play
Sep 04 12:57:33 volumio volumio[17137]: info: CoreCommandRouter::volumioPlay
Sep 04 12:57:33 volumio volumio[17137]: UNSET VOLATILE
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853657] CoreStateMachine::play index undefined
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853661] CoreStateMachine::setConsumeUpdateService undefined
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853666] CorePlayQueue::getTrack 0
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853669] CoreStateMachine::startPlaybackTimer
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853671] CorePlayQueue::getTrack 0
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853679] ControllerMpd::clearAddPlayTracks NAS/testu2/Rattle_and_Hum/03-Desire.flac
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853683] ControllerMpd::sendMpdCommand stop
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853697] sending command...
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853702] ------------------------------ 67ms
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853712] parsing response...
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853717] ControllerMpd::sendMpdCommand clear
Sep 04 12:57:33 volumio volumio[17137]: info: [1504529853719] sending command...
Sep 04 12:57:33 volumio volumio[17137]: info:
... and so on
```

Next:
 * while still playing, press 'next'
 * playback stops
 * play icon is shown
 * log file
```
Sep 04 12:58:40 volumio volumio[17137]: info:
Sep 04 12:58:40 volumio volumio[17137]: [1504529920607] ---------------------------- Client requests Volumio next
Sep 04 12:58:40 volumio volumio[17137]: info: CoreCommandRouter::volumioNext
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920629] CoreStateMachine::next
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920630] CoreStateMachine::stop
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920633] CoreStateMachine::setConsumeUpdateService undefined
Sep 04 12:58:40 volumio volumio[17137]: UNSET VOLATILE
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920637] CoreStateMachine::stPlaybackTimer
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920640] CoreStateMachine::updateTrackBlock
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920643] CorePlayQueue::getTrackBlock
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920647] CoreStateMachine::pushState
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920649] CoreStateMachine::getState
Sep 04 12:58:40 volumio volumio[17137]: info: [1504529920651] CorePlayQueue::getTrack 0
Sep 04 12:58:40 volumio volumio[17137]: info: CoreCommandRouter::volumioPushState
[... 121 lines cut ...]
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115716] No code
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115717] CoreStateMachine::pushState
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115718] CoreStateMachine::getState
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115720] CorePlayQueue::getTrack 0
Sep 04 13:01:55 volumio volumio[17137]: info: CoreCommandRouter::volumioPushState
Sep 04 13:01:55 volumio volumio[17137]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115725] interfaceApi::pushState
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115727] InterfaceWebUI::pushState
Sep 04 13:01:55 volumio volumio[17137]: info: [1504530115761] ------------------------------ 148ms
Sep 04 13:01:55 volumio volumio[17137]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/03-Desire.flac","favourite":false}
Sep 04 13:01:55 volumio volumio[17137]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/testu2/Rattle_and_Hum/03-Desire.flac","favourite":false}
```
Try playing via REST
```
$ wget -O- http://localhost:3000/api/v1/commands/?cmd=play 2>/dev/null
{"time":1504530283170,"response":"play Success"}
```
Playback starts fine. pause/play work ok in the web ui.

Try stopping - that works too.

```
$ wget -O- http://localhost:3000/api/v1/commands/?cmd=stop 2>/dev/null
{"time":1504530350782,"response":"stop Success"}
```

xipmix, 2018-09-05:
I think this could be closed.