volumio/Volumio2 #1157: REST command 'prev' after a bunch of 'next' stucks

heidou7p, 2017-05-01:
Hello,

As promissed, this is about 'prev' REST API.

Confirmed on: Raspberry Pi 3B + Volumio 2.141

**'Prev' after a bunch of 'next' stucks**

Reproduce procedure:

1. From web interface, create a queue with more than 4 songs, say a whole album. I have played an album from my SMB share. Play first song.
2. Ssh into volumio and do the following; this issues 2 'next' commands and a 'prev'.
    `for i in 1 2; do wget -O- http://localhost:3000/api/v1/commands/?cmd=next 2>/dev/null; done; wget -O- http://localhost:3000/api/v1/commands/?cmd=prev 2>/dev/null`
3. After sucessful 'next' commands, the 'prev' command wouldn't return a while. In a meantime, web interface says it is playing 2nd song but actual audio is from 3rd one. This situation timeouts in 120 seconds. Wget 'prev' returns, volumio starts playing 1st (what?!) song.

Output from 3 wget commands is like this; notice time gap between 2nd 'next' and 'prev' is about 120 seconds.

    {"time":1493608082133,"response":"next Success"}
    {"time":1493608082237,"response":"next Success"}
    {"time":1493608203568,"response":"prev Success"}

Log from 'sudo journalctl -f':
    (Notice between line 343 and 344 there's a 120 second gap)
    https://gist.github.com/heidou7p/407a7b6f8900fbd08c23254573d198f8

xipmix, 2018-09-26:
I tried to reproduce this on 2.452. It's still there.

With a 1-second pause between commands, the first try worked fine.
```
$ cat test.sh
for i in 1 2; do
wget -O- http://localhost:3000/api/v1/commands/?cmd=next 2>/dev/null;
sleep 1;
done
wget -O- http://localhost:3000/api/v1/commands/?cmd=next 2>/dev/null;

$ sh -x test.sh
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963273240,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963274358,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=prev
{"time":1537963275466,"response":"prev Success"}
```
If I reduce the delay to 0.1s, I get strange behaviour. The first test run is ok
```
$ sh -x test.sh
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963409338,"response":"next Success"}+ sleep 0.1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963409642,"response":"next Success"}+ sleep 0.1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=prev
{"time":1537963410012,"response":"prev Success"}
```
but the next shows the pause you are referring to
```
$ sh -x test.sh
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963441291,"response":"next Success"}+ sleep 0.1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537963441604,"response":"next Success"}+ sleep 0.1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=prev
{"time":1537963562785,"response":"prev Success"}
```
During the pause, the track playing did not match that shown in the playback tab.

I went back ot 1s delay, and hit the pause again.
```
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537964365282,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537964366388,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=prev
{"time":1537964488501,"response":"prev Success"}
```
I tried once more and it worked.
```
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537964749287,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=next
{"time":1537964750387,"response":"next Success"}+ sleep 1
+ wget -O- http://localhost:3000/api/v1/commands/?cmd=prev
{"time":1537964751490,"response":"prev Success"}
```
I extended the delay to 3s and also hit the pause.

So, yes, there seems to be a problem in handling this sequence of API requests. Not sure where we go from here.

xipmix, 2018-09-26:
I started trying to get this to happen while saving the journal to a file ```journalctl -f | tee test.log```; I had to wind the sleep time back to 0.1s to get it to happen reliably.

I noticed the playback title and the track actually playing got out of sync;
during the pause the playback displays the track previous to the one playing.
When the pause ends, the track previous to the one displayed is played
and displayed as the playback title (ie we jump back two tracks from what was playing during the pause).

The attached files show the log from the client and server sides of a case where we get the pause.
Note there are 3 next requests, not 2.

Trawling through I see a traceback from the kew library
```
Sep 26 13:07:33 volumio volumio[855]: info: ControllerMpd::sendMpdCommand play  
Sep 26 13:07:33 volumio volumio[855]: info: ControllerMpd::pushError            
Sep 26 13:07:33 volumio volumio[855]: info:  message=Cannot read property 'split' of undefined, stack=TypeError: Cannot read property 'split' of undefined
Sep 26 13:07:33 volumio volumio[855]: at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:290:54)
Sep 26 13:07:33 volumio volumio[855]: at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
Sep 26 13:07:33 volumio volumio[855]: at _combinedTickCallback (internal/process/next_tick.js:131:7)
Sep 26 13:07:33 volumio volumio[855]: at process._tickCallback (internal/process/next_tick.js:180:9)
Sep 26 13:07:33 volumio volumio[855]: info: sending command...                  
```
but I see the same failure on a run where the pause does not occur, so is probably an unrelated issue.

I see two 'Client requests Volumio previous' entries, one at the start of the pause and one at the end;
```
$ grep 'Client requests' server-fail.txt
Sep 26 13:07:32 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:33 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:33 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:34 volumio volumio[855]: ---------------------------- Client requests Volumio previous
Sep 26 13:09:35 volumio volumio[855]: ---------------------------- Client requests Volumio previous
```
which explains the double-hop back up the queue that I noticed.
Unfortunately, during the pause there's not a lot of useful logging.

When processing a 'next', getTrack reports on the current track number and then we see a getTrackBlock call, followed by getTrack reporting on the next track number.
It seems that something a bit different happens on 'prev' requests. On the one that pauses we get a getTrackBlock call immediately and the track number does not decrement. The following 'prev' also starts with getTrackBlock but does decrement the track number.
```
$ egrep '(Client requests|getTrack)' server-fail.txt
Sep 26 13:07:32 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:32 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 26 13:07:32 volumio volumio[855]: info: CorePlayQueue::getTrack 8
Sep 26 13:07:32 volumio volumio[855]: info: CorePlayQueue::getTrack 8
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:07:33 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: ---------------------------- Client requests Volumio next
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 11
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 11
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 11
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 11
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 11
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 11
[... 2 lines cut ...]
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 10
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 10
[... cut ...]
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 9
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 9
```

[client-fail.txt](https://github.com/volumio/Volumio2/files/2419815/client-fail.txt)
[server-fail.txt](https://github.com/volumio/Volumio2/files/2419816/server-fail.txt)

xipmix, 2018-09-26:
Staring a bit longer I see that 'next' triggers a statemachine stop but the problematic 'prev' does not.
```
Sep 26 13:07:33 volumio volumio[855]: info: CoreCommandRouter::volumioNext      
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::next              
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::stop              
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 26 13:07:33 volumio volumio[855]: UNSET VOLATILE                            
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::stPlaybackTimer    
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::updateTrackBlock  
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrackBlock        
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::pushState         
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::getState          
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10           
Sep 26 13:07:33 volumio volumio[855]: info: CoreCommandRouter::volumioPushState 
Sep 26 13:07:33 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 26 13:07:33 volumio volumio[855]: info: CoreStateMachine::serviceStop       
Sep 26 13:07:33 volumio volumio[855]: info: CorePlayQueue::getTrack 10          
Sep 26 13:07:33 volumio volumio[855]: info: CoreCommandRouter::serviceStop      
Sep 26 13:07:33 volumio volumio[855]: info: ControllerMpd::stop                 
Sep 26 13:07:33 volumio volumio[855]: info: ControllerMpd::sendMpdCommand stop  
Sep 26 13:07:33 volumio volumio[855]: info: sending command...                  
Sep 26 13:07:33 volumio volumio[855]: info:                                     
Sep 26 13:07:33 volumio volumio[855]: ---------------------------- MPD announces state update: player
```
```
Sep 26 13:07:34 volumio volumio[855]: ---------------------------- Client requests Volumio previous
Sep 26 13:07:34 volumio volumio[855]: info: CoreCommandRouter::volumioPrevious  
Sep 26 13:07:34 volumio volumio[855]: info: CoreStateMachine::previous           
Sep 26 13:07:34 volumio volumio[855]: info: CoreStateMachine::updateTrackBlock  
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrackBlock        
Sep 26 13:07:34 volumio volumio[855]: info: CoreStateMachine::pushState         
Sep 26 13:07:34 volumio volumio[855]: info: CoreStateMachine::getState          
Sep 26 13:07:34 volumio volumio[855]: info: CorePlayQueue::getTrack 10          
Sep 26 13:07:34 volumio volumio[855]: info: CoreCommandRouter::volumioPushState 
Sep 26 13:07:34 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 26 13:07:34 volumio volumio[855]: info:                                     
Sep 26 13:07:34 volumio volumio[855]: ---------------------------- MPD announces state update: player
```
The successful 'prev' does issue the stop.
```
Sep 26 13:09:35 volumio volumio[855]: ---------------------------- Client requests Volumio previous
Sep 26 13:09:35 volumio volumio[855]: info: CoreCommandRouter::volumioPrevious  
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::previous          
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::stop              
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 26 13:09:35 volumio volumio[855]: UNSET VOLATILE                             
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::stPlaybackTimer   
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::updateTrackBlock  
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrackBlock        
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::pushState         
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::getState          
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 10          
Sep 26 13:09:35 volumio volumio[855]: info: CoreCommandRouter::volumioPushState 
Sep 26 13:09:35 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 26 13:09:35 volumio volumio[855]: info: CoreStateMachine::serviceStop       
Sep 26 13:09:35 volumio volumio[855]: info: CorePlayQueue::getTrack 10          
Sep 26 13:09:35 volumio volumio[855]: info: CoreCommandRouter::serviceStop      
Sep 26 13:09:35 volumio volumio[855]: info: ControllerMpd::stop                 
Sep 26 13:09:35 volumio volumio[855]: info: ControllerMpd::sendMpdCommand stop  
Sep 26 13:09:35 volumio volumio[855]: info: sending command...                  
Sep 26 13:09:35 volumio volumio[855]: info:                                     
Sep 26 13:09:35 volumio volumio[855]: ---------------------------- MPD announces state update: player
```
Here's a diff of the logged command sequence after the first 'prev' and after the second 'prev'. It's clearer if you put them up side by side but that's not possible here.
```
@@ -1,6 +1,10 @@
  ---------------------------- Client requests Volumio previous
  info: CoreCommandRouter::volumioPrevious
  info: CoreStateMachine::previous
+ info: CoreStateMachine::stop
[... 133 lines cut ...]
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- error: error
- info:
+ info: ------------------------------ 204ms

```

xipmix, 2018-09-26:
Hmm, a bit more staring and I notice the second 'next' did not increment the track number. The traceback from the kew library occurs during the command sequence after that 'next'. But it's the only time there's such a traceback, so perhaps it's related to the track number failing to increment, perhaps not. It doesn't seem that the kew failure is the cause of the 'prev' issue.