volumio/Volumio2 #1166: Https Login

NicoHood, 2017-05-07:
It would be nice to have the web frontend available with https. This way all data is encrypted, especially the login, for example if you configure spotify. A self signed https cert would be okay. You can trust it and then the traffic is encrypted.

volumio, 2017-05-30:
That makes no sense, sorry. There is no way we can have https on local volumio, for various reasons, the first of which is, if self singned you will get a ton of warnings by every browser

closed 2017-05-30 by volumio

koenbuyens, 2019-05-20:
It would be great if 
- all the references to http:// in the code-base are replaced with protocol-relative URs (i.e. URIs starting with // instead of http:// ). 
- replacing explicit references to the IP address with the hostname of the URI
This would enable me to use nginx as reverse-proxy.

[image]

volumio, 2019-05-20:
Https over local network is not possible.

koenbuyens, 2019-05-21:
I am running https on my local network.

Koen