volumio/Volumio2 #1167: shairport-sync not autostarting on 2.163?

nilsbecker, 2017-05-07:
i recently flashed a new 2.163 image on my rpi3. most things work, but i do not see volumio appear as an airplay output device on my mac. 

when i manually run `/usr/local/bin/shairport-sync` i'm able to get volumio offered as an audio streaming target. how can i fix this so this is automatically started?

macmpi, 2017-05-08:
Could be initialization problem: did you let very first boot (after flashing) complete for about 8mn before restarting or changing anything?

nilsbecker, 2017-05-08:
i cannot say anymore. 50/50. can i copy some config files to appropriate places manually?

chsims1, 2017-05-10:
Systemd service file (/usr/lib/systemd/system/shairport-sync.service) appears to have incorrect ExecStart path.

`volumio@volumiokitchen:/lib/systemd/system$ cat /usr/lib/systemd/system/shairport-sync.service
[Unit]
Description=ShairportSync AirTunes receiver
After=sound.target
Requires=avahi-daemon.service
After=avahi-daemon.service

[Service]
ExecStart=/usr/bin/shairport-sync
User=shairport-sync
Group=shairport-sync

[Install]
WantedBy=multi-user.target
`

macmpi, 2017-05-10:
hum...
So there are duplicate services?
`/lib/systemd/system/airplay.service`
`/usr/lib/systemd/system/shairport-sync.service`

chsims1, 2017-05-10:
Ahh, that's confusing.  Airplay.service is active (starts shairport-sync).  The other one (shairport-sync.service) with the incorrect path isn't ... sorry, my mistake.

xipmix, 2017-05-10:
hint: (v2.129)
```
$ sudo systemctl status airplay.service
● airplay.service - ShairportSync AirTunes receiver
   Loaded: loaded (/lib/systemd/system/airplay.service; disabled)
   Active: active (running) since Wed 2017-05-10 10:40:29 UTC; 5s ago
 Main PID: 1321 (shairport-sync)
   CGroup: /system.slice/airplay.service
           └─1321 /usr/local/bin/shairport-sync

May 10 10:40:29 volumio systemd[1]: Started ShairportSync AirTunes receiver.
May 10 10:40:29 volumio shairport-sync[1321]: startup
```
Interesting, my service file looks like this - notice the different user.
```
$ cat /lib/systemd/system/airplay.service
[Unit]
Description=ShairportSync AirTunes receiver
After=sound.target
Requires=avahi-daemon.service
After=avahi-daemon.service

[Service]
ExecStart=/usr/local/bin/shairport-sync
User=volumio
Group=volumio

[Install]
WantedBy=multi-user.target
```

nilsbecker, 2017-05-10:
```
volumio@volumio:~$ sudo systemctl status airplay.service
● airplay.service - ShairportSync AirTunes receiver
   Loaded: loaded (/lib/systemd/system/airplay.service; disabled)
   Active: inactive (dead)
```

```
volumio@volumio:~$ cat /lib/systemd/system/airplay.service
[Unit]
Description=ShairportSync AirTunes receiver
After=sound.target
Requires=avahi-daemon.service
After=avahi-daemon.service

[Service]
ExecStart=/usr/local/bin/shairport-sync
User=volumio
Group=volumio

[Install]
WantedBy=multi-user.target
```

nilsbecker, 2017-05-10:
should i just `systemctl start airplay`?

nilsbecker, 2017-05-10:
ok i tried that -- airplay shows up. then i did `systemctl enable` and rebooted. 

seems to be working now! now only the crash after switching to the software mixer remains...

closed 2017-05-10 by nilsbecker