volumio/Volumio2 #1168: cannot set softvolume on 2.163

nilsbecker, 2017-05-07:
when setting the mixer to software volume control, the web ui becomes unresponsive. after restart, the setting for volume conrol is 'undefined'

i have to use software volume control since i have a hifiberry digi+ digital output board without hardware mixer.

nilsbecker, 2017-05-10:
this is still the same. for information, my `/etc/asound.conf` exists but is empty and my `/etc/mpd.conf` has an entry `resampler` since i need resampling to 88.2. this is working in general -- except for the software mixer.

nilsbecker, 2017-05-28:
- bump -

i am still having this issue. i now re-flashed the image 2.175 to my rpi 3.

to reproduce:

flash the image.
boot the first time.
set the sound output to hifiberry digi.  (which does not have a hardware mixer). save.
wait 10min.
reboot. 
set the mixer to software, starting volume 50%. save setting. notification: settings saved.
navigate away from the playback options page.
navigate back to playback options: the playback options page does not display.
navigate to volumio.local: the page does not display. ssh still works.
reboot.
navigate to playback options page. the mixer is set to None.

Nicolas-94, 2017-05-30:
Hello 
+1 for this, I have a quite similar issue also on 2.141 (and next ones) with Raspberry 3, and 2 (B+), and an I-sabre DAC (setting:  hifiberry DAC). Once software mixer is selected, there is no sound anymore, and volumio displays on the webpage  an error 'can not open alsa output'.  Resetting software mixer to none does not  bring back the sound on. In order to have the sound, I have to  deactivate the DAC in settings, reboot, re-activate it and reboot again. 
I did not have any such issue before 2.141
I don't have a specific asound.conf or mpd.conf setting either. 
If you need more logs, please tell me which file to put here. 

(and high 5 to volumio team for the great job)

cortegedusage, 2017-06-03:
+1 for me for this one, also with the new release 2.185 my RPI3 with i2s AUDIOPHONICS I-Sabre V3 DAC ES9023 which worked with hififerry DAC, doesn't work when i select software mixer.

Before it did work.

willing to test extra if needed.

nilsbecker, 2017-06-05:
for me also it did work before 2.163

cortegedusage, 2017-06-10:
Still no resolution?

ArwynHarris, 2017-06-10:
I have similar issue with softmixer selection making Volumio unresponsive. Problem occurred on 2.175 and is present on 2.185.    Using Pi Zero W with pHat DAC

psychophil, 2017-06-10:
Same problem as original poster. Used to be able to set volume except on Spotify plugin with Software volume control but now it just crashes when selecting software volume.
Can use Spotify Connect plugin and change volume in Spotify Android App. But that's probably not relevant.

maleg, 2017-06-13:
+1 I have same problem with raspberry pi 1B and Hifiberry Digi. Its working fine without volume control, but after activating software volume control, the web server doesnt respond and Gui is down. Ssh stil working.

heis2201, 2017-06-28:
Same symptoms as all other posters here with Durio Sound DAC, using hifiberry drivers, on a RPI B+ with Volumio 2.201

riesch, 2017-07-07:
Imo this is related to this issue: https://github.com/volumio/Volumio2/issues/1239

heis2201, 2017-07-07:
Anyone any idea for a fix or workaround?

meaz, 2017-07-17:
I have the same problem with a raspberry pi zero... It also worked before, I don't remember what Volumio version it was...

shaaati, 2017-07-26:
I am having the same problem as described by @Nicolas-94 (pHat DAC related). To give a bit of debugging help, here's my logfile: http://logs.volumio.org/volumio/XKfTxpL.html

The relevant lines are
```
Jul 26 18:31:47 x-wing volumio[837]: { Error: Command failed: /usr/bin/aplay -D softvolume /volumio/app/startup.wav
 Jul 26 18:31:47 x-wing volumio[837]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM softvolume
Jul 26 18:31:47 x-wing volumio[837]: aplay: main:722: audio open error: No such file or directory
Jul 26 18:31:47 x-wing volumio[837]: at ChildProcess.exithandler (child_process.js:206:12)
Jul 26 18:31:47 x-wing volumio[837]: at emitTwo (events.js:106:13)
Jul 26 18:31:47 x-wing volumio[837]: at ChildProcess.emit (events.js:191:7)
Jul 26 18:31:47 x-wing volumio[837]: at maybeClose (internal/child_process.js:877:16)
Jul 26 18:31:47 x-wing volumio[837]: at Socket. (internal/child_process.js:334:11)
Jul 26 18:31:47 x-wing volumio[837]: at emitOne (events.js:96:13)
Jul 26 18:31:47 x-wing volumio[837]: at Socket.emit (events.js:188:7)
Jul 26 18:31:47 x-wing volumio[837]: at Pipe._handle.close [as _onclose] (net.js:498:12)
Jul 26 18:31:47 x-wing volumio[837]: killed: false,
Jul 26 18:31:47 x-wing volumio[837]: code: 1,
Jul 26 18:31:47 x-wing volumio[837]: signal: null,
Jul 26 18:31:47 x-wing volumio[837]: cmd: '/usr/bin/aplay -D softvolume /volumio/app/startup.wav' }```

The error message when trying to play a song states the same:
'Error
Failed to open "alsa" [alsa]; Failed to open ALSA device "softvolume": No such file or directory
Error'

I don't know anything about alsa internals but somehow the change to software mixer causes volumio to think it can use a device "softvolume" that doesn't exist.

antiprism, 2017-08-05:
I am also affected by this. I posted the details to the forum here
[https://volumio.org/forum/cannot-set-software-mixer-t7231.html#p36091](url)

luckynrslevin, 2017-08-05:
Could you test my fix and send feedback?
Details see: https://github.com/volumio/Volumio2/pull/1288#issuecomment-320468485

antiprism, 2017-08-06:
I used the upgrade button to bring the version up to 2.246, and everything now appears to be working correctly. I selected "Mixer Type: Software", and this set "Mixer Control Name: Software Volume Control", and the GUI didn't hang. I can change the volume control from 100%, and this changes the listening volume.

I therefore haven't applied your changes, but thank you for looking into the problem.

luckynrslevin, 2017-08-06:
Hi @antiprism 
yes, according to my tests in my setup (RPI3 and HifiBerry DAC) this only works once. If I now change e.g. the "Default Startup Value" to any other value and save it, I will end up in an alsa error message, when trying to play music. This should be solved by my fix.

cortegedusage, 2017-08-06:
But does it still work after reboot?
And does it work with Spotify plugin and YouTube plugin?

Last time I checken It only worked for mpd and stopped working after a
reboot.

On hollidays right now without Pi so cannot check, but will do as soon I,m
home.

antiprism, 2017-08-06:
I had "Default Startup Volume" set to "disable", and was able to reboot and the software volume still worked. I changed the "Default Startup Volume" to 50 and saw alsa errors. I changed the setting back to disable and rebooted and still had alsa errors. I will apply your changes and let you know.

luckynrslevin, 2017-08-06:
@antiprism This exactly matches the behavior in my setup. Be aware that the fix is not designed to recover a system, where the configuration is messed up.
To avoid reinstallation, you could try to apply my fix, then switch back to Mixer Type "None" reboot and switch to "Software" again, this maybe recovers the configuration on your current system. If not you will have to do flash your sd card as written in my procedure (see link above).

antiprism, 2017-08-06:
After applying your changes playback was working again. I was able to change "Default Startup Volume" without errors, and on reboot the volume was set to the value and everything appeared to be working correctly. Thanks.

I do not use the Spotify or YouTube plugins and so have not tested these.

luckynrslevin, 2017-08-06:
@antiprism Thank you for testing.

Nicolas-94, 2017-08-07:
@luckynrslevin 
Here is my setup : Rasp 3, + I-Sabre DAC (driver : hifiberry DAC) (+ tactile LCD 7'', and IR receiver, but this is probably not pertinent here)
I followed your procedure, and I'm glad to confirm that this works as well on my side : volume control is working OK with standard MP3/Flac files, Youtube  or WebRadios playing, even after reboot, as well as startup volume setting. 
I don't use spotify (no account), so I could not test with this.
Anyway, Thank you very much for your work on this fix. 
Nicolas-94

meaz, 2017-08-07:
I works for me too! I have a pi zer and 2-3 pluggins. I also followed your procedure @luckynrslevin Thanks a lot!

luckynrslevin, 2017-08-08:
@Nicolas-94 @meaz thank you for testing and the appreciation. I opened a pull request to get the fix merged to the master branch (see above).

psychophil, 2017-08-08:
I installed the latest dev version of volumio2 today and set volume control
to software and all seems to be working even after a restart.

luckynrslevin, 2017-08-08:
@psychophil  Try setting the "Default Startup Volume" to another value push the save button and see what happens afterwards ...

closed 2018-03-26 by volumio