volumio/Volumio2 #1182: Add to playlist doesn't add subfolders tracks

Arthurin, 2017-05-17:
Everything is in the title! Using last release of Volumio on Rpi, tracks are stored on an USB stick.

liberatorcnx, 2017-05-17:
Can you give more information?  I tried adding various folders to a playlist, including my 'Various Artists' album folder and it added every track from every sub-folder (30+ albums) to a playlist.

Here's what I did:
Created a playlist with all of the tracks from an album.

Browsed to Albums, then added the folder 'Various Artists' to the playlist I created above - works fine and added all tracks from all 'Various Artists' albums to the playlist.

Browsed by Artist add added all albums by Artist X to playlist - works fine and added all of Artist X albums and tracks to the playlist.

If you can describe exactly what you did maybe someone can help.

Arthurin, 2017-05-17:
Sure let's detail the steps :
- Browse > Music Library > USB > select specific folder (containing multiple albums) options > Add to playlist > playlist named "Soundtracks"
[image]
No error during the operation and I even got the toast message telling me it worked, but when I look at the playlist it's empty, or even better I checked the websocket communication
[image]
As you can see the pushAddToPlaylist is empty ?!

liberatorcnx, 2017-06-02:
Hmm - it seems that this feature has never been written.  In the relevant section of playlistManager.js it says  'TODO: Deal with folders' .....  It is not dissimilar to adding to queue which does work so it may not be too difficult.  I'm not sure if anyone is already working on this?  @volumio ?

volumio, 2017-06-02:
Wanna help on this?

liberatorcnx, 2017-06-02:
@volumio - Are you talking to me?  I will have a look...  No promises (I can only work synchronously) !!!!

volumio, 2017-06-02:
Then you might want to dig into promises... It's a hell at first sight, but after it its just fantastic...

liberatorcnx, 2017-06-02:
Sorry - you missed my joke @volumio -  Irish humour (No promises means I don't guarantee (promise) I can fix it).  Anyway I will have a look and see if I can come up with a workable solution.

volumio, 2017-06-02:
Ahahaha now that you explained, I got it, and it was good and nerdy, as I like it... xkcd style...

liberatorcnx, 2017-07-04:
@volumio - this issue can be closed.

closed 2017-07-04 by volumio