volumio/Volumio2 #1184: Installing libasound2-dev fails in VERSION: 2.175

Labzz, 2017-05-20:
Hi, 

I noticed the same issue tryed to install snapcast server + client

After flashed volumio VERSION: 2.175 on my Raspberry Pi 2B i tryed everything above and a few more fixes, some broke the entire volumio image so I restarted from scratch. 

Commands I used for installing snapcast can be found [https://github.com/badaix/snapcast/blob/master/doc/build.md#linux-native](here).
 
Another thing to notice is that installing for example 'git and build-essentials' goes well and the installation steps from snapcast worked well on my Pi 3 running raspbian Jessy lite but not on the Volumio image.

[image]

Okay, I figured out that I can simply change libasound2-dev to libasound2, and yes, the installation will run correctly. 
But when i run "make" in de snapcast folder, the following appears. It seems like the libary will be added in the "dev" install. 
[image]

volumio, 2017-05-22:
I will include libasound's new version on Volumio in the next release so you won't have such problem...

Labzz, 2017-05-22:
Great thanks, can you tell me something about the new update release date?

Labzz, 2017-05-22:
Maybe it works better when I try it to installing and configuring from: https://github.com/volumio/Build
Instant of the Preinstalled image. I'll try to test it later this week.

volumio, 2017-06-02:
Done

closed 2017-06-02 by volumio