volumio/Volumio2 #1185: Plugin uninstall.sh is not executed during plugin uninstallation.

heidou7p, 2017-05-21:
Plugin's uninstall.sh script never get executed despite what told in document (https://volumio.github.io/docs/Plugin_System/Writing_A_Plugin.html); oh, wait. It never says it will be executed, but I believe that's what it meant to be.
(By the way, install.sh is SH(1) script, not a BASH(1) script as doc says. I believe uninstall.sh is the same.)

My future plugin will install some system stuff. I want to remove them by uninstallation, but it won't because uninstall.sh doesn't get executed.

Current app/pluginmanager.js has a function PluginManager.executeInstallationScript() but there's no PluginManager.executeUninstallationScript(). 

Currnet flow in PluginManager.unInstallPlugin() is:
	stopPlugin()
	disablePlugin()
	removePluginFromConfiguration()
	pluginFolderCleanup()

Probably we should run unistall.sh between removePluginFromConfiguration() and pluginFolderCleanup() ?

macmpi, 2017-05-21:
duplicate of https://github.com/volumio/Volumio2/issues/851

volumio, 2017-06-02:
done https://github.com/volumio/Volumio2/commit/c21785b0877305c02ee26eb8fce7c92443d19e4c

closed 2017-06-02 by volumio