volumio/Volumio2 #1186: albums with the same name

Warter21, 2017-05-21:
https://volumio.org/forum/albums-with-the-same-name-t5719.html

liberatorcnx, 2017-05-21:
You will also notice that only the first found album is listed when you view albums.  So, if you have 5 albums called 'Greatest Hits' only one will be shown when you view albums.  If you select that album to view the tracks then the tracks from ALL 'Greatest Hits' albums are shown.

It happens because the search is only by album name (no artist specified) and only unique records are returned when you view all albums.  When you view tracks it finds all tracks on any album called 'Greatest Hits'.

I already fixed this in my PR 1170 which is pending.

Warter21, 2017-05-21:
Unfortunately it is in the album view. You can see all "Greatest Hits" albums and if you go into any of these albums you can see all "Greatest Hits" tracks on it. No mather who is the arist.

liberatorcnx, 2017-05-21:
This is one of the problems that is fixed in PR 1170.  It will now show the tracks only from the album and artist requested.  Current version shows tracks from matching album names only, so it finds all those tracks

Warter21, 2017-05-21:
Thats good! I am waiting for the merge. Thanks a lot!

closed 2017-06-02 by volumio