volumio/Volumio2 #1189: UPnP renderer outputs high frequency noise

michaelosthege, 2017-05-22:
Hi,
I am running Volumio 2 on a Raspberry Pi Model B Rev 2 and my objective is to use it as a UPnP renderer, with the 3.5 mm jack as the default output.
I can play webradio via the web UI just fine.

However when I try to "Play To" from my Windows 10 laptop, it __only plays high frequency components__ of the audio stream. The track and album art are shown in the UI as expected.
The `upmpcli.conf` all at default settings.

Before Volumio I tried to use LibreELEC and the UPnP renderer worked fine, however I also need to run a Python background service, which is not possible on LibreELEC. Also Volumio seems more responsive.

volumio, 2017-06-07:
Should have been fixed with this : https://github.com/volumio/Volumio2/commit/c363d66bd8dddb334e627dc88713b83a6984e18a

michaelosthege, 2017-06-08:
I tried to pull from master, but after reboot the volumio service won't start up.
I can ssh into it, but to me it looks like the renderer has problems:

`
systemctl --failed
  UNIT             LOAD   ACTIVE SUB    DESCRIPTION
<E2><97><8F> upmpdcli.service loaded failed failed UPnP Renderer front-end to MPD
`

The `volumio start` commands takes about half a second and gives no feedback. `volumio vstop` and `volumio restart` fail with `/usr/local/bin/volumio: line 134: stop: command not found`.

The current state is that I can't reach the web UI.

volumio, 2017-06-08:
The fix has been done on the OS, so volumio pull won't help.
If you have ssh access still do 

touch /boot/factory_reset 
sync
reboot

to reinitiate a factory reset. 

If you want to try the fix, flash this image instead:
http://updates.volumio.org/pi/volumio/2.188/volumio-2.188-2017-06-08-pi.img.zip

michaelosthege, 2017-06-10:
I flashed that image, but the renderer is still choking on the StreamWhatYouHear stream. I've tuned into a webradio and it briefly stops playback but then resumes the webradio. The Windows "Cast to device" tells me that the "device is not responding".
That's all I have for the moment.

volumio, 2017-06-10:
Ok I'll try stream what you hear. How do you use the cast t device function on win?

michaelosthege, 2017-06-10:
Right click a mp3 file and then here should be a „Play to“ or „cast to device“ option.

closed 2017-11-30 by volumio