volumio/Volumio2 #1190: Alarm setting cannot be disabled

cldebuhr, 2017-05-22:
I've recently started using the alarm function with Volumio. Works great. Set the time, set to "on", and select the desired playlist, and the system begins playing the specified playlist at the specified time. Perfect. The problem is when I don't want the alarm (i.e., weekends, etc.). Set to "off", go to sleep, and it still plays at the specified time! Am I wrong in assuming the on/off toggle is there to temporarily disable the alarm?

This issue has been present since at least 2.141 on a setup consisting of a Rpi2 with a HiFiBerry Dac+ (standard) and a USB WiFi dongle. Target playlist consists of a single webradio stream. For unrelated reasons (firmware upgrade troubles - since resolved), I flashed a new firmware image on several occasions and the problem persisted. Furthermore, I've recently deployed 2.175 on brand-new hardware (Rpi3/Allo Boss - internal WiFi) and the problem is present on this system as well. 

If any additional info/logfiles/etc. are required to track this down let me know and I'll do what I can

Bond246, 2017-05-25:
i have the same problem. I'm using volumio just for 6 weeks now, it never worked to disable the alarm. I always had to delete it to prevent calling me up.

volumio, 2017-06-02:
Fixed with https://github.com/volumio/Volumio2/commit/f2c64530a332af220a634ede5fd7b12c4d9b1811

closed 2017-06-02 by volumio