volumio/Volumio2 #1195: v2.175: HifiBerry DAC+ doesn't work on RPi3

youpilai, 2017-06-01:
Hi,
I've done an update from an old version (0.9XX) on my RPI3 setup with Hifiberry DAC+, and couldn't make the sound working on the DAC anymore.
I've tried to make a fresh install of the last version, v2.175, telling me that updating from a 0.9XX version was quite a jump, but with no more results (i've also tried with an older release from december, 2.038 or something like that, with no luck)

When activating I2S, selecting the Hifiberry DAC Plus entry in the DAC list, and selecting this output, no sound is coming from Volumio. I've checked the audio jack output and the hdmi one, everything is ok.

I noticed that i also missing the hardware mixer option, i only have "none" or "software" entry, and selecting software doesn't work.

I've finally tried another software (RuneAudio) to check my hardware, and everything is working fine, so i suppose the problem is coming from Volumio
As i'm surely not the only one to use a RPi3 with a Hifiberry DAC+ here, perhaps this problem has already been refered to, but i didn't found any mention of it.

volumio, 2017-06-01:
Did you try to reboot?

youpilai, 2017-06-01:
yes, i tried to reboot, multiple times. I know it often solves a lot of things 
(it's my third volumio install: i own 
1 RPI Zero w/ IQaudio DAC, 
1 RPi2 w/ Durio Sound Pro,
and 1 RPi3 w/ HifiBerry DAC+,
)

M-I-Kessler, 2017-06-06:
It works if Output Device is set to Hifiberry DAC and I2S DAC off.  If I2S is on and set to Hifiberry DAC Plus sound will out put an error. Work around select Hifiberry DAC Plus and save, then on the Output Device select Hifiberry DAC Plus and It should work properly.

heis2201, 2017-06-27:
I have a similar problem but I am posting it here since @youpilai owns similar hardware.
My problem is with a Durio Sound Pro on a RpiB+ using the Hifiberry driver. I know this DAC has no hardware volume control, but Software control doesn't work. Volumio becomes unresponsive and I have hard-reset it.

Any ideas for a fix or workaround here?

closed 2018-03-26 by volumio