volumio/Volumio2 #1197: HIFIBERRY DIGI+ PRO does not work on Raspberry Pi 2 B

mecworks, 2017-06-02:
I have a Raspberry Pi 2 B (40pin header, quad core) and a newly purchased Hifiberry DIGI+ Pro.  No matter what HifiBerry DAC I select (The Hifiberry DIGI+ Pro is NOT available as an actual choice), I cannot get it to work at all.  I reboot after each change when testing the DACs to no avail.

Volumio version: 2.175
Date: 16-05-2017

volumio, 2017-06-02:
Known issue, we need to update the kernel

closed 2017-06-02 by volumio

mecworks, 2017-06-03:
Do you have an ETA for a release that has a new kernel that will work with the Hifiberry DIGI+ Pro?

psychophil, 2017-06-08:
Also waiting for comparability with volumio2 before purchasing a pro version board.

Using a digi+ with pi3 and Volumio2 and volspotconnect2 plugin with Chord Mojo DAC.