volumio/Volumio2 #1201: Sound distorted for open.fm radio

michalwu, 2017-06-04:
Volumio v 2.185
Hifiberry DAC +

Most of web radios working fine. But when I add stream from open.fm, I clearly hear sound is garbled. There is no audio problem when I open same stream via browser or android app. I tried reboot, factory settings...

Radio is open.FM, using AAC+ (HE-AAC v2) @ 48 kbps
Stream example: 
http://gr-relay-lb.gaduradio.pl/20

Logs: http://logs.volumio.org/volumio/zL4hhAK.html

michalwu, 2017-06-27:
Just short update.
Volumio v 2.201 is the same.

michalwu, 2017-09-26:
v2.246 and no change. Is there anyone who is willing to solve this? Please let me know what else I can provide to finally close this issue.

michalwu, 2017-10-30:
v2.296 and nothing changed. Sound is still distorted and volume low.
I created new log bundle: http://logs.volumio.org/volumio/VR3wYwN.html
Please help!

volumio, 2017-10-31:
Can you please try to enable resampling to 24 bits and 192 khrtz? Let me know

michalwu, 2017-10-31:
Thanks. Changed to 24bit, 192k. I rebooted volumio just to be sure. After reboot new settings were there. Openfm playback was still garbled. I tried some other radio -it was fine.
Streaming url from first post is broken, here is latest one: http://stream.open.fm/20?

New logs uploaded. Have no idea if it helps.
http://logs.volumio.org/volumio/cRyxsDW.html

michalwu, 2018-11-21:
Year later. Latest volumio (fresh install) and same issue with http://stream.open.fm/20. Logs here: http://logs.volumio.org/volumio/38DVprs.html

I found solution (by michal12121212 from elektroda), but need help with verification if it won't break anything else. Is this plugin needed in Volumio?
Just edit /etc/mpd.conf and add:
decoder { 
plugin "faad" 
enabled "no" 
} 

Now openfm streams working fine. Logs: http://logs.volumio.org/volumio/pL9pFRK.html

closed 2020-04-09 by michalwu