volumio/Volumio2 #1203: Duplicated Album Art Sever

b1tdust, 2017-06-05:
Hi, I'm running Volumio2 on a RPi B+. Everything is working pretty well. As my RPi has only ~500mb of RAM I'm constantly checking the ps output and I found this

> 44.812500 Mb /usr/local/bin/node /volumio/app/plugins/miscellanea/albumart/serverStartup.js 3001 /data/albumart 
>  51.082031 Mb /bin/node /volumio/app/plugins/miscellanea/albumart/serverStartup.js 3001 /data/albumart 

I have only the youtube plugin. Is that ok? How can I disable the album art server?

xipmix, 2017-06-06:
Can you post your logs (go to /dev url of volumio, click on 'send logs'.) This will provide a lot more information that will help to debug the issue.

There is an option in the My Music menu item that disables albumart lookup, I am not sure it does precisely what you want.

volumio, 2017-06-06:
Not an issue. We spawn several instances of the albumart server to speed up albumart fetching (number of cpu cores -1 )

closed 2017-06-06 by volumio

b1tdust, 2017-06-07:
Thanks for the quick reply!
Would it cause an issue on volumio service if I change the code on [serverStartup.js](https://github.com/volumio/Volumio2/blob/65a94466ffb36c11c7284dc41b9a7a51af9ea419/app/plugins/miscellanea/albumart/serverStartup.js) to run one or either none instance of it? I tried to disable it in the **plugins.json** but didn't work. Any safe way to disable it?