volumio/Volumio2 #1206: SambaCry -> Update Samba

z-n-g-g-u-v-n-f, 2017-06-07:
[CVE-2017-7494](https://www.samba.org/samba/security/CVE-2017-7494.html) discribes a remote code execution based on Samba.

There is a tool on Gitlab, called [SamaCry](https://github.com/Waffles-2/SambaCry) to check if a system is vulnerable and
**Volumio version 2.185 is vulnerable!**

To reproduce it:
(Make sure you haven [nmap](https://nmap.org) installed.)

```
git clone https://github.com/Waffles-2/SambaCry.git
cd SambaCry
nmap --script=./CVE-2017-7494.nse volumio.local
```

My output is like this:

```
$ nmap --script=./CVE-2017-7494.nse volumio.local

Starting Nmap 7.40 ( https://nmap.org ) at 2017-06-07 12:36 CEST
Nmap scan report for volumio.local (192.168.226.28)
Host is up (0.0068s latency).
rDNS record for 192.168.226.28: volumio.fritz.box
Not shown: 989 closed ports
PORT      STATE SERVICE
22/tcp    open  ssh
80/tcp    open  http
111/tcp   open  rpcbind
139/tcp   open  netbios-ssn
445/tcp   open  microsoft-ds
3000/tcp  open  ppp
3001/tcp  open  nessus
3005/tcp  open  deslogin
3006/tcp  open  deslogind
5000/tcp  open  upnp
49152/tcp open  unknown

Host script results:
| CVE-2017-7494: 
|   State is: POTENTIALLY VULNERABLE
|   Samba-vuln-CVE-2017-7494
|   Summary: Remote code execution from a writable share.
|   Description: A Samba vulnerability (CVE-2017-7494) enables a malicious attacker with valid write access to a file share to upload and execute an arbitrary binary file which will run with Samba permissions.
|   Affected Version: All versions of Samba from 3.5.0 onwards.
|_  For more info: https://www.guardicore.com/2017/05/samba/

Nmap done: 1 IP address (1 host up) scanned in 5.49 seconds
```

I think updating Samba should fix this remote code execution.

volumio, 2017-06-13:
I'm just waiting for raspbian to update the package...

volumio, 2017-06-14:
Seems rasbpian updated the package, and we're not vulnerable any more 

volumio@volumio:~/SambaCry$ nmap --script=./CVE-2017-7494.nse volumio.local

Starting Nmap 6.47 ( http://nmap.org ) at 2017-06-14 12:16 UTC
Nmap scan report for volumio.local (192.168.1.11)
Host is up (0.0029s latency).
rDNS record for 192.168.1.11: volumio.station
Not shown: 990 closed ports
PORT      STATE SERVICE
22/tcp    open  ssh
111/tcp   open  rpcbind
139/tcp   open  netbios-ssn
445/tcp   open  microsoft-ds
3000/tcp  open  ppp
3001/tcp  open  nessus
3005/tcp  open  deslogin
3006/tcp  open  deslogind
5000/tcp  open  upnp
49154/tcp open  unknown

Host script results:
| CVE-2017-7494:
|   os: Windows 6.1
|   lanmanager: Samba 4.2.14-Debian
|   server: VOLUMIO\x00
|   date: 2017-06-14T12:16:07+00:00
|   fqdn: localhost
|   domain_dns: \x00
|_  workgroup: WORKGROUP\x00

Nmap done: 1 IP address (1 host up) scanned in 0.92 seconds

closed 2017-06-14 by volumio

z-n-g-g-u-v-n-f, 2017-08-04:
Updating to System Version 2.224 this is still not fixed.