volumio/Volumio2 #1208: I2s dac wrong outputdevice number

balbuze, 2017-06-08:
In a fresh install 2.186
After enabling loopback device, the outputdevice value is wrong for the dac.
aplay -l gives
`aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: IQaudIODAC [IQaudIODAC], device 0: IQaudIO DAC HiFi pcm512x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`
but no sound because in /data/configuration/audio_interface/alsa_controller/config.json 
`{
  "volumestart": {
    "type": "string",
    "value": "disabled"
  },
  "volumemax": {
    "type": "string",
    "value": "100"
  },
  "volumecurvemode": {
    "type": "string",
    "value": "logarithmic"
  },
  "outputdevice": {
    "type": "string",
    "value": "1"
  },
  "mixer_type": {
    "type": "string",
    "value": "Hardware"
  },
  "mixer": {
    "type": "string",
    "value": "Digital"
  },
  "volumesteps": {
    "type": "string",
    "value": "10"
  },
  "outputdevicename": {
    "type": "string",
    "value": "IQaudIO DAC Plus"
  }
}
`
Even after reboot, save, change for jackoutput. But sometimes it works. As result no sound.

balbuze, 2017-06-08:
seems to be the same problem than https://github.com/volumio/Volumio2/issues/1195

M-I-Kessler, 2017-06-09:
Did the work around work for you?

balbuze, 2017-06-09:
yes !

balbuze, 2017-06-11:
To be more precise, on a fresh install, it works. But if snd_aloop is loaded, and i2s activated, then the outputdevice number is wrong, and all the system is configured with this wrong value with no sound as result.

volumio, 2017-06-11:
We should add options snd-aloop index=8" >> /etc/modprobe.d/alsa-base.conf

This way we will have the loopback module always in 8th position and nothing will break

balbuze, 2017-06-11:
ok thanks ! I'll test it !

volumio, 2017-06-13:
How do you load snd aloop?

balbuze, 2017-06-13:
In install.sh modprobe snd_aloop to use without reboot and write in /etc/modules for next boot. Plus now write in /etc/modprobe.d/alsa-base.conf snd_aloop index=7

volumio, 2017-06-13:
Does it work?

balbuze, 2017-06-13:
Yes it seems ok. Loopback is loaded with the good number index. But I met some other problems. I'll ask you in a other thread

volumio, 2017-06-13:
Ok, however the ideal situation is that we put the mod in etc/modprobe.d/alsa-base.conf from the build of the image. If you write on this with the plugin, it can create some problems in the future.
I am looking at it now

closed 2017-11-16 by volumio