volumio/Volumio2 #1214: playplaylist REST command doesn't work with curl on RPI3 v2.193

cortegedusage, 2017-06-11:
When I try to load a playlist with:
curl http://volumio.local:3000/api/v1/commands/?cmd=playplaylist&name=FIP
it reports:
{"time":1497168594269,"response":"playplaylist Success"}
but playing doesn't start
when giving the command throug a browser, it does play, and it reports the same...

other commands given with curl ?cmd=play or stop do work.
it doesn't matter if I'm giving the curl command from the rpi itself or from an other rpi...

liberatorcnx, 2017-06-12:
Try:
curl 'http://volumio.local:3000/api/v1/commands/?cmd=playplaylist&name=FIP'

liberatorcnx, 2017-06-12:
Ohh - also worth mentioning:  You must URL encode any name with unsafe ASCII characters (spaces etc).

curl 'http://volumio.local:3000/api/v1/commands/?cmd=playplaylist&name=AC DC' will not work.
curl 'http://volumio.local:3000/api/v1/commands/?cmd=playplaylist&name=AC%20DC' will work fine.

cortegedusage, 2017-06-12:
Ok rtfm for me
Thank you so much for this!

closed 2017-06-12 by cortegedusage