volumio/Volumio2 #1217: kernelsource: silentoldconfig fails due to new config symbols

taudac, 2017-06-11:
`sudo volumio kernelsource`  fails, see below:

```
[-- snip --]
scripts/kconfig/conf  --silentoldconfig Kconfig
*
* Restart config...
*
*
* Display Panels
*
support for simple panels (DRM_PANEL_SIMPLE) [N/m/?] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
Makefile:542: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
make: *** No rule to make target 'modules_prepare'.  Stop.
[-- snip --]
```

Seems like the downloaded kernel sources do not correspond to the running kernel.

`uname -r` returns:

    4.4.9-v7+

`head /usr/src/rpi-linux/Makefile` returns:

```
VERSION = 4
PATCHLEVEL = 4
SUBLEVEL = 50
EXTRAVERSION =
NAME = Blurry Fish Butt
[-- snip --]
```

volumio, 2017-06-11:
Any idea on how to fix it?

taudac, 2017-06-11:
@volumio I was hoping you could tell me... :smile:

taudac, 2017-06-11:
@volumio Where do you get `/boot/.firmware_revision` from?

taudac, 2017-06-11:
Changing the content of `/boot/.firmware_revision` to [`15ffab5493d74b12194e6bfc5bbb1c0f71140155`](https://github.com/Hexxeh/rpi-firmware/commit/15ffab5493d74b12194e6bfc5bbb1c0f71140155) solved the issue.

@volumio  Seems like `/boot/.firmware_revision` does not correspond to the running kernel, what do you think? Can we fix it?

xipmix, 2017-06-12:
There is a program in debian called 'uscan' that can be quite helpful for checking if there is a new upstream version of some package. It might be useful for the build script to run it and fail if there is a new version. It would require the build env to have the 'devscripts' package installed but it would not need to go into the built image.

volumio, 2017-06-12:
I think this commit created this bug:
https://github.com/volumio/Build/commit/5e379a94149eb26f1dc493a83236a9c85006a94a

@macmpi any comment on this? 
We need to have the firmware revision to match the kernel revision to make kernelsource.sh to work

@xipmix what's the point of making the script fail if there is a new version?

macmpi, 2017-06-12:
hum, strange...
it is a `rpi_update` [thing](https://github.com/Hexxeh/rpi-update/blob/master/rpi-update#L231) that is supposed to be created in any case if I read correctly.

volumio, 2017-06-12:
Yes, but with the commit I referenced we have the  firmware revision not matching the kernel one...

macmpi, 2017-06-12:
I see. so we might just need to save file after kernel install and restore after firmware update

xipmix, 2017-06-12:
@volumio - the failure brings attention to the new situation (the build log is quite voluminous). There are other possible courses of action of course.

volumio, 2017-06-12:
IMHO we can leave this as it is, given that we will switch kernel in a week (where firmware commit is equal to kernel commit)

macmpi, 2017-06-12:
@volumio don't you think that in the future, we may still encounter cases (though not desirable) where firmware and kernel commit # may have to be different ?
Will there be a discipline to readily update kernel as soon as a new device appear for instance, (though kernel may be the same and just firmware changes)?

Maybe we should fix it now rather than be sorry, if at some point one make some ad-hoc discrepancy between these commits.

in [kernelsource.sh](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/volumio_command_line_client/commands/kernelsource.sh) we may just care about a `/boot/.kernel_revision` file that is a renamed `/boot/.firmware_revision` right after kernel's first `rpi-update` command in `raspberryconfig.sh`.

volumio, 2017-06-12:
That is really a good solution, agree on all the line. You can proceed with that if you want

taudac, 2017-06-12:
@volumio I think this commit is the root cause for this bug: [volumio/Build/cac931ed](https://github.com/volumio/Build/commit/cac931eda81b99ff610cf0e9252dffb61388371a)

The [first `rpi-update`](https://github.com/volumio/Build/blob/cac931eda81b99ff610cf0e9252dffb61388371a/scripts/raspberryconfig.sh#L65) installs kernel 4.4.9 and the corresponding firmaware, the [second `rpi-update`](https://github.com/volumio/Build/blob/cac931eda81b99ff610cf0e9252dffb61388371a/scripts/raspberryconfig.sh#L68) updates the firmware to the latest revision. Now the firmware and the kernel revisions do not match any more. Is this intended? Can we revert this commit for the next release?

macmpi, 2017-06-12:
@volumio fine.
Won't be able to do it for a few days unfortunately: anyone may feel free to do as proposed if needed earlier.

volumio, 2017-06-13:
Done with: 
https://github.com/volumio/Build/commit/a8b1e012aca57474aa4e2f90c5ab63590525f578
https://github.com/volumio/Volumio2/commit/db8ea1d00f1b18e9e47fdb9549b3cc0cc88a8e92

It will be available in next release

closed 2017-06-13 by volumio

volumio, 2018-09-11:
Is there something more to do?