volumio/Volumio2 #1228: Have netbios and mdns support

shawaj, 2017-06-15:
MoOde audio player has both netbios and mDNS support. Meaning you can use http://moode, moode.local or IP address to access the interface.

This is extremely important - as on Windows and most Android devices mDNS does not work out of the box but netbios does!

Please can you add this same functionality?

volumio, 2017-06-15:
Sure we can do it, and actually did not know that netbios could work on Android. 
I am  researching the matter now, but didn't find anything conclusive, do you have any resource I can take inspiration on?

shawaj, 2017-06-15:
Maybe look into how MoOde do it?

francesco-vannini, 2017-06-15:
Android is a pain in the back for name resolution. Some vendors implemented mDNS whereas others not, the DNS setting sent via DHCP is often ignored and NetBIOS is a bit of a mystery in terms of native support but there are definitely apps that implement it.
Would it at all make sense to develop a small app the likes of ZeroConf Browser
https://play.google.com/store/apps/details?id=com.melloware.zeroconf&hl=en_GB
which returns the Volumio/JustBoom installations in the local network and allow to open a web browser by clicking on them? This could also be a nice companion for the setup of Volumio when starting in AP mode.

If anything though NetBIOS is good for the Windows platform so that people don't need to install Apple Bonjour.

volumio, 2017-06-28:
Did a bit of investigation and we have both mdns and netbios enabled. http://volumio works . Could you report more info on the failing system?

francesco-vannini, 2017-06-28:
I must admit I didn't realise SAMBA was enabled on Volumio and I guess I was sidetracked by the fact that I was checking thing on the JustBoom Player. Sorry about this.
I looked into this further and it would seem that for the JustBoom Player the NetBIOS name in smb.conf is set as "Justboom Player". The space is the problem. The list here https://support.microsoft.com/en-us/help/909264/naming-conventions-in-active-directory-for-computers-domains-sites-and provides the allowed characters and despite the space isn't explicitly mentioned this article explains why it is nevertheless not allowed http://windowsitpro.com/networking/why-cant-i-have-spaces-my-windows-2000-netbios-domain-name
Changing the smb.conf to "justboomplayer" or "justboom" fixes things.
Taking this further I have tried to set the NetBIOS name to justboom.local which then matches the mDNS type of name to see if that could improve things when trying to access the player from different platforms. This might also be a good way forward for volumio.
So far avahi-resolve returns the right answer from mDNS and on a Windows box nbtstat -a x.y.z.w also returns the NetBIOS name I set as justboom.local
All in all this is encouraging, maybe this issue wasn't opened for nothing after all :)

volumio, 2017-06-28:
You're right, the space is causing the issue, will fix on JustBoom player, what is your favourite name convention?

francesco-vannini, 2017-06-28:
maybe for the moment we could keep it as "justboom". If, as I am hoping, this approach that I have just described above works then we can change it to "justboom.local". Could you give me a few days to test things further? At least we make one change only. It would also be nice to hear what you guys think about it too. If this works it is pretty much the only distro to adopt an approach to naming that would work on almost every platform.

shawaj, 2017-07-11:
I was testing this in the office today, it appears justboom.local does not work as windows does not recognise justboom.local 

Bizarrely, if I ping 192.168.1.253 which is the IP of the player locally, it gives me a reverse resolution of justboom.local, but then using that in a browser or in ping, tracert etc it does not work. I guess because the TLD confuses it and tries to resolve the DNS, which fails. This is on a machine which does not have itunes/bonjour installed at all.

With simply justboom in /etc/samba/smb.conf it does work. See full outputs below, but ultimately I think we will have to go with something with no punctuation.

For justboom.local:

```
nbtstat -a 192.168.1.253

Ethernet:
Node IpAddress: [0.0.0.0] Scope Id: []

    Host not found.

Bluetooth Network Connection:
Node IpAddress: [0.0.0.0] Scope Id: []

    Host not found.

WiFi:
Node IpAddress: [192.168.1.121] Scope Id: []

           NetBIOS Remote Machine Name Table

       Name               Type         Status
    ---------------------------------------------
    JUSTBOOM.LOCAL <00>  UNIQUE      Registered
    JUSTBOOM.LOCAL <03>  UNIQUE      Registered
    JUSTBOOM.LOCAL <20>  UNIQUE      Registered
    WORKGROUP      <00>  GROUP       Registered
    WORKGROUP      <1E>  GROUP       Registered

    MAC Address = 00-00-00-00-00-00

Local Area Connection* 2:
Node IpAddress: [0.0.0.0] Scope Id: []

    Host not found.

Ethernet 2:
Node IpAddress: [0.0.0.0] Scope Id: []

    Host not found.

Local Area Connection* 14:
Node IpAddress: [0.0.0.0] Scope Id: []

    Host not found.
```

```
ping justboom.local
Ping request could not find host justboom.local. Please check the name and try again.
```

```
ping 192.168.1.253

Pinging 192.168.1.253 with 32 bytes of data:
Reply from 192.168.1.253: bytes=32 time=9ms TTL=64
Reply from 192.168.1.253: bytes=32 time=11ms TTL=64
Reply from 192.168.1.253: bytes=32 time=12ms TTL=64
Reply from 192.168.1.253: bytes=32 time=9ms TTL=64

Ping statistics for 192.168.1.253:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 9ms, Maximum = 12ms, Average = 10ms
```

```
tracert 192.168.1.253

Tracing route to JUSTBOOM.LOCAL [192.168.1.253]
over a maximum of 30 hops:

  1    10 ms    13 ms     8 ms  JUSTBOOM.LOCAL [192.168.1.253]

Trace complete.
```

```
tracert justboom.local
Unable to resolve target system name justboom.local.
```

Then changing it to just justboom:

```
tracert justboom

Tracing route to justboom [192.168.1.253]
over a maximum of 30 hops:

  1    46 ms     4 ms     3 ms  JUSTBOOM [192.168.1.253]

Trace complete.
```

```
ping justboom

Pinging justboom [192.168.1.253] with 32 bytes of data:
Reply from 192.168.1.253: bytes=32 time=6ms TTL=64
Reply from 192.168.1.253: bytes=32 time=6ms TTL=64
Reply from 192.168.1.253: bytes=32 time=6ms TTL=64
Reply from 192.168.1.253: bytes=32 time=7ms TTL=64

Ping statistics for 192.168.1.253:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 6ms, Maximum = 7ms, Average = 6ms
```

francesco-vannini, 2017-07-11:
I guess that if we can live with having "justboom" for NetBIOS and "justboom.local" for Avahi then we could easily settle for that. We know it works, we just have to properly inform the users to which one to use depending on their platform.

shawaj, 2017-07-11:
@francesco-vannini does NetBIOS not work on all systems then?

xipmix, 2017-09-04:
After reading this discussion I noticed this passage [in the MS reference](https://support.microsoft.com/en-us/help/909264/naming-conventions-in-active-directory-for-computers-domains-sites-and)

> The use of non-DNS names with periods is allowed in Microsoft Windows NT. However, periods should not be used in Microsoft Windows 2000 or in later versions of Windows.

This may be helpful for #974, thanks all.

francesco-vannini, 2017-09-04:
You are right @xipmix but from what I have seen it is largely tolerated. The problem obviously roots into the . being a separator within active directory.

> Names can contain a period (.). However, the name cannot start with a period. The use of non-DNS names with periods is allowed in Microsoft Windows NT. However, periods should not be used in Microsoft Windows 2000 or in later versions of Windows.

It is a **should** more than a **must** and the good thing of it all is that it works when implemented. I think having only one name type to refer to independently of which resolution mechanism one uses could be really good for users which more often than not don't care too much about naming resolution.

shawaj, 2017-09-05:
When I tested it, justboom.local didn't work on NetBIOS. As soon as I changed it to JustBoom only it worked. Just FYI.

closed 2017-11-16 by volumio

shawaj, 2017-11-16:
@volumio did this get fixed then?