volumio/Volumio2 #1232: Airplay 2

andre-dw, 2017-06-19:
Will Volumio 2 support Airplay 2 in the future?

volumio, 2017-06-28:
No idea since nothing is known about the new protocol and its advantages over airplay1

closed 2017-06-28 by volumio

snikcers, 2017-08-23:
Multiroom support! Just to name the biggest advantage!

benjaminwood, 2018-09-13:
Worth noting the discussion going on over here: https://github.com/mikebrady/shairport-sync/issues/535

I too would love airplay 2 for multiroom support. I'm sure somebody is working to reverse engineer the protocol. Perhaps when details are shared this issue can be reopened?

Neustradamus, 2019-12-08:
@volumio: Can you reopen this ticket or I need to create a new?

NickBNY, 2020-01-21:
airplay2 is key feature https://www.google.com/amp/www.iphonehacks.com/2019/12/airplay-2-cracked-stream-audio-unsupported-speakers.html/amp

volumio, 2020-01-21:
There is still nothing that can be implemented.

acresp, 2021-07-08:
@volumio AirPlay 2 is now supported in the dev branch of [shairport-sync](https://github.com/mikebrady/shairport-sync)
Do you think it would be possible to implement it in a beta or something?

kimdiallo, 2021-09-24:
AirPlay2 works already on HifiBerryOS (dev branch) - maybe interesting:

 https://github.com/hifiberry/hifiberry-os/blob/668fdf8b72ffa8c87875ac856e12c83725b7ce77/buildroot/package/hifiberry-shairport/hifiberry-shairport.mk

Would love to get this feature also from Volumio!