volumio/Volumio2 #1233: mpd 0.20.6-1 bug - not showing all NAS folders

MicheleM79, 2017-06-21:
Background as following:

Volumio latest release for RPi.

1. LAN with a Ubuntu home server, music folder shared with samba, reachable from windows pc, smartphones, tablets, and clearly from Volumio.
2. The only folder that does not appear is "F" (under "music" folder there's a folder for every letter, with respective artists subfolders), as following:

../share/music/A/...
../share/music/B/...
....
../share/music/Z/...
every letter folder contains respective artists.
No permissions problems, no corrupt flacs.
The first time I configured the NAS source in Volumio the "F" folder did not appear. So I went with "UPDATE MPD DB", with no other actions: the folder then appeared and the subfolders (only two) were visible, but no flac files where listed inside. Strangely though the "F" folder appeared at the end of the list ("A", "B", "C",....."Z", "F").

I can confirm that a new version of MPD (0.20.9 with a new config that disables the outdated and buggy sndfile and audiofile codecs and only uses ffmpeg) solves the problem, completely.
Thanks

volumio, 2017-06-21:
What is the the configuration changes you ve made to mpd 20.9

MicheleM79, 2017-06-21:
Thanks for reply.
Unfortunately I don't have the time to do this type of testing, and I based my claim on the testing made by the author of Moode Player, with my collaboration. I tried this system, and the behavior was the same identical of Volumio. So the author send me a binary of mpd 0.29 (as described in the previous message) and the problem was immediately gone. Next official release will then include this version of mpd. I know it's another system/player, but considering the behavior, I bet the problem is the same. 
Is it possible to take care of it? I would really, really enjoy the x86 version :)
Thanks

volumio, 2017-06-22:
Can you tell me (maybe via mail) what is the mpd.conf that you tried? Send to info at volumio dot org thanks so much!

MicheleM79, 2017-06-22:
Hi (Michelangelo? Gé?... :) )

here I am: attached you can find the binary of mpd 0.20.9, the one that
solves the problem.
And also the mpd.conf, copied from Moode Player.

Do you think it's possible to solve the problem in Volumio? It would be so
nice!

Thanks, ciao
Michele

volumio, 2017-06-23:
thank you michele, I'll look at it next week, and sure we can solve it!

mi

MicheleM79, 2017-06-23:
Hi Michelangelo,

many thanks, I can't wait to try the new version.

Ciao 
Michele

volumio, 2017-06-27:
Grazie a te!!!

closed 2017-11-16 by volumio