volumio/Volumio2 #1237: Audio shuttering after update to 2.201

pilifsch, 2017-06-26:
Hi guys,

I have a problem after last update. I updated Volume from something about v2.110 (can't remember exactly) to the newest version and I am experiencing audio shuttering every time while playing over airplay. This is not happening while playing webradio.
I tried increasing buffer size to 12 MB and fiddling with other options (DSD playback, audio resampling) but with no luck. Can you please help me figure out what is wrong?

Thanks,
Philip

pilifsch, 2017-07-11:
Anybody can help with resolving this issue?

jamesw4, 2017-07-17:
Same issue here aswell.

volumio, 2017-07-23:
Guys can you please share more infos about your setup?

pilifsch, 2017-07-23:
Sure I can.
As for hardware part. I have RPi 1 model B with WiFi USB dongle and external DAC connected through USB too. RPi connects to my home network using WiFi only. Power supply is efficient (2.4 A).
OS that I use is of course Volumio OS. However I have installed additional software like Homebridge and Mosquitto for my home automation.

I should mention that I have complete backup of state before update (created image of SD card). By that I can reproduce this issue. Let me know if we can use it somehow to find the cause of this problem.

empossible, 2017-07-23:
I've got the same issues (crackling, popping, disconnecting entirely). Mine is on a fresh install of 2.201 after formatting the SD card.  I did this twice with the same results.  I'm using a RPi 3 over WiFi, with a CanaKit power supply.  

After a third format, I tried the Volumio Hotspot.  That has resulted in to issues so far (about 40 minutes of playback).  When I switched back to my WiFi network, all the issues immediately return.

I would say it's an issue with my WiFi network, but I've also run RuneAudio on the same device and it had no trouble with Airplay over my WiFi network.

volumio, 2017-07-23:
Can you please add:
audio_backend_buffer_desired_length_in_seconds = 0.30;

in the alsa section of /volumio/app/plugins/music_service/airplay_emulation/shairport-sync.conf.tmpl

and reboot? does it help?

empossible, 2017-07-23:
Ok, I rebooted from the WebUI and the same issue arose immediately.  I tried again a few minutes later without rebooting, and I received static followed by a crash and reboot.  After the crash and reboot boot, it worked!  For like, 30 seconds.  Subsequent reboots resulted in the static, popping, and audio dropping immediately on playback.

Just to verify, my _shairport-sync.conf.tmpl_ now looks like this:

> alsa =
{
  output_device = "${device}";
  audio_backend_buffer_desired_length_in_seconds = 0.30;
};

I also receive speaker popping on shutdown and startup, but I don't think that's related.

volumio, 2017-07-23:
Can you please send us logs?

empossible, 2017-07-23:
[Log Upload](http://logs.volumio.org/volumio/tLQYFhp.html)

volumio, 2017-07-24:
What I mean is: please send us logs after the problem occurs

empossible, 2017-07-24:
Sorry about that, here's one directly after the issue occurs.  [Log Upload](http://logs.volumio.org/volumio/YXSw3Xh.html)

Just to note, I let the issue run for about 30 seconds, and I took the log without interacting with the playback.

volumio, 2017-07-24:
That's very helpful, thanks. What I see is:
Error sendto-ing to audio socket: Invalid argument

Could you please send the content of /etc/shairport-sync.conf ?

thanks

empossible, 2017-07-25:
```
general =
{
    name = "Volumio";
    log_verbosity = 0;
    interpolation = "soxr";
};

alsa =
{
  output_device = "plughw:0,0";
  audio_backend_buffer_desired_length_in_seconds = 0.30;
};

sessioncontrol =
{
  allow_session_interruption = "yes";
  run_this_after_play_ends = "/usr/local/bin/volumio stopairplay";
};

metadata =
{
    enabled = "yes";
    include_cover_art = "no";
    //pipe_name = "/tmp/shairport-sync-metadata";
    //pipe_timeout = 5000;
    socket_address = "226.0.0.1";
    socket_port = 5555;
};

```

pilifsch, 2017-07-29:
Sorry for such long delay with my answering.
I did update Volumio from version 2.141 to 2.224 but it can't boot successfully after that. I see it connected to my home network, but can't access it through WebUI or even ssh (Connection refused).
Let me know if you have any idea how to diagnose it (make a log).

empossible, 2017-07-30:
Update on my end:

Updated to 2.245, which has almost 100% corrected the issue.  Now, I get very, very rare popping from playback.  More prevalent though, is that every other song or so, the audio will drop out for about 10 seconds and then continue on.  I tried to grab an error log, but both attempts have locked up at "Sending log report, please wait."

I personally don't care about this, but I also noticed, that Volumio is not displaying track information or album art in the WebUI via Airplay.

pilifsch, 2017-07-30:
Update from my side:

I restored my "backup" image of Volumio and tried updating it once again. I did successful update to 2.224 and then 2.245. After that AirPlay works better, but unfortunately shuttering still occurs. It happens every 10 seconds, sometimes 15 or 20... :( 
I tried sending log but I cannot send it successfully. I it takes forever and I still see "Sending log report, please wait" message. Anyways I will try to upload it.

xipmix, 2017-07-30:
You may be able to find out where it is getting stuck. 
First, you have to re-enable SSH login (go to http://volumio/dev)
Then log in and try to send a log. Shortly after you press the button, try these commands
```
$ tail /tmp/logondemand
$ tail /var/log/volumio.log
```
These will display the last 10 lines of each file. That should show at least the command before the one that gets stuck, and possibly the one that is getting stuck. If you could paste them here that would be useful too.
My guess is that it is the final upload that gets stuck. To check for that
```
$ ps -fade|grep curl
```

pilifsch, 2017-07-31:
Here is the output:

```
root@dobrebity:/home/volumio# tail /tmp/logondemand
root       107     2  0 18:43 ?        00:00:01 [loop0]
root       110     2  0 18:43 ?        00:00:00 [kworker/0:1H]
root       150     2  0 18:44 ?        00:00:00 [jbd2/mmcblk0p3-]
root       151     2  0 18:44 ?        00:00:00 [ext4-rsv-conver]
root       158     2  0 18:44 ?        00:00:00 [ipv6_addrconf]
root       179     1  0 18:44 ?        00:00:04 /lib/systemd/systemd-journald
root       183     2  0 18:44 ?        00:00:00 [kworker/u2:2]
root       205     1  0 18:44 ?        00:00:00 /lib/systemd/systemd-udevd
root       290     2  0 18:44 ?        00:00:00 [cfg80211]
root       386     2  0 18:44 ?        00:00:0

root@dobrebity:/home/volumio# tail /var/log/volumio.log
2017-07-31T18:53:36.275Z - info: [1501527216269] ------------------------------ 105ms
2017-07-31T18:53:36.327Z - info: CoreCommandRouter::volumioGetQueue
2017-07-31T18:53:36.334Z - info: [1501527216328] CoreStateMachine::getQueue
2017-07-31T18:53:36.342Z - info: [1501527216336] CorePlayQueue::getQueue
2017-07-31T18:53:36.354Z - info: 
[1501527216344] ---------------------------- Client requests Volumio queue
2017-07-31T18:53:36.358Z - info: [1501527216356] InterfaceWebUI::pushQueue
2017-07-31T18:53:36.364Z - info: [1501527216362] ------------------------------ 42ms
2017-07-31T18:53:38.700Z - info: CALLMETHOD: system_controller system sendBugReport [object Object]
2017-07-31T18:53:38.707Z - info: CoreCommandRouter::executeOnPlugin: system , sendBugReport

root@dobrebity:/home/volumio# ps -fade|grep curl
root      3827  3290  0 18:54 pts/0    00:00:00 grep curl
```

pilifsch, 2017-07-31:
And again, after reboot and trying to play audio through AirPlay for few minutes:

```
root@dobrebity:/home/volumio# tail /tmp/logondemand
Mon Jul 31 19:18:37 UTC 2017

root@dobrebity:/home/volumio# tail /var/log/volumio.log
2017-07-31T19:19:48.705Z - info: [1501528788698] CoreStateMachine::syncState
2017-07-31T19:19:48.714Z - info: [1501528788706] CoreStateMachine::pushState
2017-07-31T19:19:48.733Z - info: [1501528788715] CoreStateMachine::getState
2017-07-31T19:19:48.742Z - info: CoreCommandRouter::volumioPushState
2017-07-31T19:19:48.751Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2017-07-31T19:19:48.771Z - info: [1501528788755] interfaceApi::pushState
2017-07-31T19:19:48.784Z - info: [1501528788773] InterfaceWebUI::pushState
2017-07-31T19:19:49.054Z - info: Pushing Favourites {"service":"airplay","uri":"","favourite":false}
2017-07-31T19:19:49.094Z - info: Pushing Favourites {"service":"airplay","uri":"","favourite":false}
2017-07-31T19:19:49.137Z - info: Pushing Favourites {"service":"airplay","uri":"","favourite":false}

root@dobrebity:/home/volumio# ps -fade|grep curl
root      3466  3348  0 19:20 pts/0    00:00:00 grep curl
```

xipmix, 2017-08-01:
So it looks like the log sending module completed this command:
```
ps -ef
```
but took a long time to do the final task:
```
sudo journalctl -p 7
```
The log does get cleaned up a little but yours is not so large that should take minutes...

Ah. Here we are.
```
Jul 31 19:18:33 dobrebity volumio[962]: info: CALLMETHOD: system_controller system sendBugReport [object Object]
Jul 31 19:18:33 dobrebity volumio[962]: info: CoreCommandRouter::executeOnPlugin: system , sendBugReport
Jul 31 19:18:38 dobrebity volumio[962]: info: Canot send bug report: Error: Command failed: /usr/local/bin/node /volumio/logsubmit.js "Audio shuttering (Airplay) - issue #1237"
Jul 31 19:18:38 dobrebity volumio[962]: /bin/sh: 1: Syntax error: "(" unexpected
Jul 31 19:18:38 dobrebity volumio[962]: /volumio/logsubmit.js:31
Jul 31 19:18:38 dobrebity volumio[962]: console.log(error)
Jul 31 19:18:38 dobrebity volumio[962]: ^
Jul 31 19:18:38 dobrebity volumio[962]: ReferenceError: error is not defined
Jul 31 19:18:38 dobrebity volumio[962]: at Object. (/volumio/logsubmit.js:31:17)
Jul 31 19:18:38 dobrebity volumio[962]: at Module._compile (module.js:570:32)
Jul 31 19:18:38 dobrebity volumio[962]: at Object.Module._extensions..js (module.js:579:10)
Jul 31 19:18:38 dobrebity volumio[962]: at Module.load (module.js:487:32)
Jul 31 19:18:38 dobrebity volumio[962]: at tryModuleLoad (module.js:446:12)
Jul 31 19:18:38 dobrebity volumio[962]: at Function.Module._load (module.js:438:3)
Jul 31 19:18:38 dobrebity volumio[962]: at Module.runMain (module.js:604:10)
Jul 31 19:18:38 dobrebity volumio[962]: at run (bootstrap_node.js:389:7)
Jul 31 19:18:38 dobrebity volumio[962]: at startup (bootstrap_node.js:149:9)
Jul 31 19:18:38 dobrebity volumio[962]: at bootstrap_node.js:504:3
```
The string you entered with the log report is being interpreted by the shell when logsubmit.js is called.
I'll work up a PR to fix that.

As for your _actual_ problem, I am not sure. I suspect wifi interference. If you are able to use a wired network link, does the problem go away?

pilifsch, 2017-08-01:
Unfortunately I can't test it through wired connection because my network AP is in totally different location. I will try sharing wideo network from my laptop which will also access network through WiFi. I will try to do that and let you know in about 8 hours.

Anyways I feel that it shouldn't be WiFi interface fault because I tried two APs (now I have a new, much better one) and and WiFi dongle used with RPi is based on popular chipset that is for sure 100% supported by OS. And moreover this issue occurs only after those updates. Before update it worked perfect all the time.

xipmix, 2017-08-01:
Ok, thanks for clearing up the wifi angle, which was a guess at best.

pilifsch, 2017-08-01:
Shuttering occurs in intervals around between 10-30 seconds. I will try to collect few logs but before this I would like to clear if I understand it correctly.

1) I will reboot Volumio
2) Log in using SSH
3) Run: 
`volumio$ tail -f /var/log/volumio`
4) Play music using AirPlay
5) Stop it (ctrl+c) immediately when I hear a shutter
6) Collect last minute of this log using example:
`volumio$ sudo journalctl --utc --since="2017-08-01 10:03" --until="2017-08-01 10:04"`
7) What do I do then? Copy and paste it here?

xipmix, 2017-08-01:
I realised something that will make this easier.
Your general understanding is correct, but we can make steps 6 & 7
easier for you. Just use the 'send a log' feature _but_ when you
enter the description, avoid any of these characters:

 ```!$(){}[]``` or backslash.

The log send should then work just fine.

pilifsch, 2017-08-02:
Hello again. I was not able to run 'tail -f /var/log/volumio', needed to add '.log' at the end.
Unfortunately I couldn't get any data to upload because:
`volumio@dobrebity:/var/log$ journalctl
No journal files were found.`

Upload of log took forever and did't upload anything at all. No matter what title I used to upload (no special characters).

xipmix, 2017-08-02:
Sorry for the mistake in the log file name.
Another mistake, journalctl needs sudo prefixed, i.e.
```
  $ sudo journalctl ... etc
```
 
> Upload of log took forever and did't upload anything at all.
> No matter what title I used to upload (no special characters).

Hm. That's surprising. I will try to reproduce.

xipmix, 2017-08-04:
Well this is a head-scratcher. Somehow the data being fed to 
```http/dev/dev.js:socket.on('pushSendBugReport, function(data)```
is being corrupted. I added these three lines to that function
```
+    console.log('pushSendBugReport start data:');                               
+    console.log(data);                                                          
+    console.log('pushSendBugReport end data:');                                 
     var json = JSON.parse(data);
```
and in the JS console of the browser debug tool I get
```
pushSendBugReport start data:
description: 'simple test 3'
/usr/bin/curl -X POST -H 'Content-Type: multipart/form-data' -F 'logFile=@/tmp/logondemand' -F desc='simple test 3' -F 'variant=volumio' 'http://logs.volumio.org:7171/logs/v1'
{"status":"OK","link":"http://logs.volumio.org/volumio/Ig63ngo.html"}
pushSendBugReport end data:
SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data
```
so somehow stdout is getting cleared before logsubmit.js emits its result. This is causing JSON.parse to explode. We could clean up the input data before trying to parse it but this makes me wonder why this ever worked.

xipmix, 2017-08-05:
Even more mysterious. I added some code to strip anything before or after the ```{ }``` delimiting the json. When I first tested, the data was back to being ok (ie no stripping required) and the report URL was displayed as intended. Removed all the changes to dev.js, still working, report url displayed ok.

Once while I was poking and prodding I got an error on the JS with wording _something_ like:
```clearMessageStr not defined```
This coincided with the behaviour @pilifsch described - the 'submitting log' message appears but nothing happens thereafter; in the system log logsubmit.js is never called. Unfortunately I didn't make a proper note of the message text. For now I've added some preparsing of the data in my next PR.

xipmix, 2017-08-05:
While I was poking and prodding I got an error on the JS:
```ReferenceError: clearBrowseView not defined```
This appeared when I had the '/dev' page open with a log url displayed,
and then put /dev in the url bar of the browser and hit <return>.

The error traceback showed this reference to dev.js
```
<anonymous> http://<volumioip>/dev/dev.js:45:2
```
That line in dev.js is part of ```socket.on('disconnect', function() {``` and reads
```
44    clearPlayQueue();                                                           
45    clearBrowseView();                                                          
46    clearPlaylistView();                                                        
```
With the browser in this state I once produced the behaviour @pilifsch described - the 'submitting log' message appeared but nothing happens thereafter; in the system log logsubmit.js was never called. But only once. The next time I tried to reproduce this behaviour the code worked as expected.
If it matters, this was with FF54 on ubuntu 16.04.

pilifsch, 2017-08-05:
@xipmix, sorry for late reply. I will try to collect the log again tomorrow. Till then, let me know if there is something more that I can do to help you track and fix those issues.

pilifsch, 2017-08-07:
Hi again. I don't know which part I am doing wrong but I cannot get anything that might be useful for you. Three things below.

1) Log upload through web service still does not work for me.

2) '--utc' option is still not recognised by 'sudo journalctl'; anyways I managed to get this kind of output:
```
root@dobrebity:/home/volumio# sudo journalctl --since="2017-08-07 22:30"
-- Logs begin at Mon 2017-08-07 21:55:47 UTC, end at Mon 2017-08-07 22:30:57 UTC. --
Aug 07 22:30:35 dobrebity sudo[8491]: volumio : TTY=pts/1 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/su
Aug 07 22:30:35 dobrebity sudo[8491]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Aug 07 22:30:35 dobrebity su[8518]: Successful su for root by root
Aug 07 22:30:35 dobrebity su[8518]: + /dev/pts/1 root:root
Aug 07 22:30:35 dobrebity su[8518]: pam_unix(su:session): session opened for user root by volumio(uid=0)
Aug 07 22:30:51 dobrebity sudo[8609]: root : TTY=pts/1 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl --utc --since=2017-08-07 22:30
Aug 07 22:30:51 dobrebity sudo[8609]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Aug 07 22:30:51 dobrebity sudo[8609]: pam_unix(sudo:session): session closed for user root
Aug 07 22:30:57 dobrebity sudo[8659]: root : TTY=pts/1 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl --since=2017-08-07 22:30
Aug 07 22:30:57 dobrebity sudo[8659]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
```
But anyways I see that it will not help a lot...

3) I did everything by following the "plan" that was written before and I assume that there should be something valuable in journalctl...

4) Please guide me what can I do more to help in any way.

xipmix, 2017-08-08:
Thanks for trying. We can forget about --utc, the output shows the timezone anyway. I played around and something like this is probably the way to go:
```
$ sudo journalctl --since="-5m"
```
which will display the last five minutes of log data.

But, as you show above, even if we can show the logs, they lack useful clues to the stuttering problem.
I've never used shairport or airplay so I am shooting in the dark here but there are a few obvious things to try.
 * From the [manual](http://htmlpreview.github.io/?https://github.com/mikebrady/shairport-sync/blob/master/man/shairport-sync.html) it seems you can set
```
log_verbosity=3
```
in the config file (```sudo nano /volumio/app/plugins/music_service/airplay_emulation/shairport-sync.conf.tmpl```) and reboot. This should produce more (possibly too much) debug info from the shairport daemon, which might help.

 * From the [troubleshooting page](https://github.com/mikebrady/shairport-sync/blob/master/TROUBLESHOOTING.md) it looks like disabling wireless power management might be useful, but from the log you posted that seems to be in place already.

 * the troubleshooting page specifically mentions Volumio2 does not have an asound.conf but says it puts configuration in /etc/mpd.conf. The file distributedin Volumio2 has not been modified in some time so it's hard to see where the change of behaviour comes from. I think mpd has had a version bump, perhaps there is need for a corresponding tweak to the mpd.conf file.

 * Can we see your /etc/shairport-sync.conf, to compare with @empossible ? And  /etc/mpd.conf, and if you have one, /etc/asound.conf ? It looks like the output of ```aplay -l``` might be useful too.
To suppress the comment lines, which aren't active configuration, try
```
$ grep -v '^#' /etc/shairport-sync.conf
$ grep -v '^#' /etc/mpd.conf 
```

My guess is you will end up solving this by tweaking the stream bandwidth to be in the [CD quality range](https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md#knownissues), or the sampling parameters which probably involves editing /etc/mpd.conf. Which parameters, we don't know yet. We need some information on what rates volumio is trying to achieve.

 * One last avenue to try is tweaking shairport as was done here: mikebrady/shairport-sync#517 . Note this was with an older version of volumio (May 2017) so I have no idea if those changes will help.

pilifsch, 2017-08-13:
Here you go:

**shairport-sync.conf**
```
general =
{
    name = "dobrebity";
    log_verbosity = 0;
    interpolation = "soxr";
};

alsa =
{
  output_device = "plughw:5,0";
};

sessioncontrol =
{
  allow_session_interruption = "yes";
  run_this_after_play_ends = "/usr/local/bin/volumio stopairplay";
};

metadata =
{
    enabled = "yes";
    include_cover_art = "no";
    //pipe_name = "/tmp/shairport-sync-metadata";
    //pipe_timeout = 5000;
    socket_address = "226.0.0.1";
    socket_port = 5555;
};
```

**mpd.conf**
```
music_directory		"/var/lib/mpd/music"
playlist_directory		"/var/lib/mpd/playlists"
db_file			"/var/lib/mpd/tag_cache"

user				"mpd"
group                          "audio"
bind_to_address		"any"
gapless_mp3_playback			"no"
auto_update    "yes"
follow_outside_symlinks	"yes"
follow_inside_symlinks		"yes"
input {
        plugin "curl"
}

audio_output {
		type		"alsa"
		name		"alsa"
		device		"softvolume"
		dop			"no"
		
		

}

audio_output {
    type            "fifo"
    enabled         "no"
    name            "multiroom"
    path            "/tmp/snapfifo"
    format          "44100:16:2"
}

volume_normalization		"no"

audio_buffer_size		"8192"
buffer_before_play		"10%"

max_connections			"20"
max_playlist_length		"32384"
max_command_list_size		"8192"
max_output_buffer_size		"16384"

filesystem_charset		"UTF-8"
id3v1_encoding			"UTF-8"

```

pilifsch, 2017-08-13:
**aplay -l**
```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 5: S51 [SB X-Fi Surround 5.1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 5: S51 [SB X-Fi Surround 5.1], device 1: USB Audio [USB Audio #1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

pilifsch, 2017-08-13:
Unfortunately my volumio broken (probably) after setting 'log_verbosity=3' and it won't boot anymore. Let me know if this log is still needed because I would need to restore my backup, run through updates but I don't have time for that right now. Probably tomorrow or other day later this week.
Waiting for info.

Thank you so much for spending so much time on my case!

xipmix, 2017-08-13:
> Unfortunately my volumio broken (probably) after setting
> 'log_verbosity=3' and it won't boot anymore. Let me know if this
> log is still needed because I would need to restore my backup,
> run through updates but I don't have time for that right now.

Very strange. I'll see if I can reproduce the breakage.
Thanks for all the info so far.

xipmix, 2017-08-14:
I changed log_verbosity to 3 and had no issue, so I'm not sure why volumio won't boot anymore.
Best to try with a fresh 2.246 image, configure it just enough to test with (network, ssh, one music library). Test with that, save the log:
```
$ sudo bash
# journalctl > /var/tmp/test1.log
# scp /var/tmp/test1.log me@laptop_ip_addr
```
Then try upping the shairport log_verbosity (no other changes) and reboot. Test with that, save the log.

Then we can try going further. In the 'Playback Options' screen, what 'Output Device' is selected ?
The reason I ask is that I see in your report above you have softvolume selected in mpd.conf but shairport-sync.conf says the output is to your USB audio device - I don't know if that will matter but it seems not quite right to me.
Since you will have a fresh system, things may be different than above, so check what the web UI shows is selected, then check the config files.

If you find them to be the same as above and want to try editing the mpd.conf to make it consistent with shairport-sync.conf, I would try replacing the audio_output section with this:
```
audio_output {
  type "alsa"
  name "alsa"
  device "hw:5,0"
}
```
(best to comment out the old audio_output section so you can revert the change if necessary)
and then restart volumio to force mpd to reread the file
```
$ sudo systemctl restart volumio
```
I have no real understanding of the implications here so if anyone thinks this is a bad idea, speak up!

There are a couple of issues with softvolume that are being worked on, see #1300. Perhaps this problem is another aspect of those issues.

I-alx-I, 2017-08-24:
I had these problems too, but deleting the line 5 '`interpolation = "soxr";`' in ../shairport-sync.conf.tmpl (as in the current github version) and reboot works for me.

volumio, 2017-08-24:
I confirm that removing soxr can be the trick. I've done it today, expect it in next release

closed 2017-11-30 by volumio