volumio/Volumio2 #1239: Software mixer not working: RpiB+ Durio (hifiberry driver) on v2.201

heis2201, 2017-06-27:
I would love to use Software volume control (even though I loose some quality) but I can't get it working on Volumio 2.201.

Configuration:

> Output device: Hifiberry DAC
> I2S DAC: ON
> DAC model: Hifiberry DAC
> Mixer type: none/Software

When I set Mixer type to Software, and then Save, Volumio crashes and I have to power-off the raspi.

Anyone suggestions? Thanks!

volumio, 2017-06-27:
Which raspberry pi model do you have?

heis2201, 2017-06-27:
Sorry, forgot to mention: It's a B+

heis2201, 2017-06-27:
I have found the following information in my kernel log: What does it mean, and is it related?

```
[  +0.794184] snd_bcm2835: unknown parameter 'index' ignored
[  +0.013552] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  +0.069926] snd-hifiberry-dac soc:sound: ASoC: CPU DAI (null) not registered
[  +0.008998] snd-hifiberry-dac soc:sound: snd_soc_register_card() failed: -517
[  +0.034636] sd 1:0:0:0: Attached scsi generic sg1 type 0
[  +0.148353] snd-hifiberry-dac soc:sound: ASoC: CPU DAI (null) not registered
[  +0.009035] snd-hifiberry-dac soc:sound: snd_soc_register_card() failed: -517
```

nilsbecker, 2017-06-28:
seems related to #1168 (although that is on rpi3 and started in 2.163)

riesch, 2017-07-07:
Exactly the same problem and behaviour with raspberry pi zero w and hifiberry miniamp. 
Output device: Hifiberry DAC
I2S DAC: ON
DAC model: Hifiberry DAC
Mixer type: none/Software

riesch, 2017-07-07:
Imo this is related to https://github.com/volumio/Volumio2/issues/1168

heis2201, 2017-07-07:
as @nilsbecker mentioned [above](https://github.com/volumio/Volumio2/issues/1239#issuecomment-311775552)...

heis2201, 2017-07-10:
anyone any ideas how to fix this?

heis2201, 2017-08-30:
I assume we can close this bug since https://github.com/volumio/Volumio2/pull/1300 has been merged into master (https://github.com/volumio/Volumio2/pull/1300#issuecomment-324387222)

closed 2017-08-30 by heis2201