volumio/Volumio2 #1240: Sleep mode not work

suypportedox, 2017-06-28:
Hello

Thanks for this wonderfull works !! 

i juste have a little problem, the sleep mode not work on my RPi3, i use the .img install

if i do a sleep, i have the confirmation pop-up shows " NaN " ...
[image]

Thanks

xipmix, 2017-06-28:
Can you tell us what version this is please. Look in the 'system' page of right-side menu.

suypportedox, 2017-06-28:
Version is 2.201   from  volumio-2.201-2017-06-13-pi.img.zip

xipmix, 2017-06-28:
This seems to be a cosmetic issue. Does it actually go to sleep when you want it to?

suypportedox, 2017-06-28:
ok the sleep mode work and if i do just 1 hour , i have not the NaN but replace by will ... 
it is just a cosmetic so.

but i have to sleep at a time fixed not with timing ... it is possible ,

closed 2017-07-04 by volumio