volumio/Volumio2 #1241: Schedule Radio

suypportedox, 2017-06-28:
Hello

i search to do a Schedule for web radio.

Just play a web radio at 8am stop 12am and replay 14pm and restop18pm

The Pi3 is always on .

It is possible ??

2lepus, 2017-12-23:
There is an alarm clock functionality, which is able to play Webradio if you put the Station to a playlist.
I tried this without success with a former Version, but with the current release it works. :-)
But there is still no possibility to stop by schedule

2lepus, 2017-12-23:
When trying to add Songs to the playlist from a Smartphone I am not allways able to select from the context menu.
The context menu is allways shown above. When I choose the first entry, the context menu exeeds the Display. And while the context menu is shown I am not able to scroll, so the menu is unreachable.
On tablets the context menu is shown below when clicking upper entries.

relft, 2021-01-16:
Can you give detail to ho put a webradio to a playlist to shcedule an alarm?
Thanks