volumio/Volumio2 #1243: Feature suggestion: Installer / Way to input wifi details during installation

BLX-S, 2017-06-28:
Would it be an idea to ship volumio with a flash tool, kind of like osmc, where you specify your network details and everything else and it creates flashes the image. 

I just want a way to specify wifi without having to plug my pi into LAN every time I install volumio. This would be handy for Pi Zero W... For example PiMusicBox has a config file on the SD-Card which you can add your wifi credentials to by just using a text editor when your sd-card is plugged into the computer.

chsims1, 2017-06-30:
You can add your wifi credentials to '/etc/wpa_supplicant/wpa_supplicant.conf' on the sd card image.

volumio, 2017-06-30:
That's the wrong way to do it, I advise you against doing it

BTW: You can set up your wifi from the hotspt

chsims1, 2017-06-30:
I should have added a caveat that it's not the preferred way (presumably because it potentially messes up updates?? or is there another reason?). Yes, ofc it can be done from the hotspot (and I do), but that wasn't the OP request.

macmpi, 2017-06-30:
Actually for Pi Zero (non-W and W), I was thinking about enabling it's convenient ethernet gadget mode, which may help for initial setup (plug device over USB onto another machine and get local network from that machine and do initial setup through normal web UI).

It's just a few minor config file changes to do, but would need to be tested on other Pis to make sure it does not alter normal USB operations with those...

BLX-S, 2017-06-30:
Thanks guys, I wasn't aware of the hotspot feature until I saw it yesterday but the temporary suggestions are helpful till the whole installer thing gets off the ground.

closed 2017-11-16 by volumio