volumio/Volumio2 #1244: Blank web interface: SyntaxError: expected expression, got '<'

dup2, 2017-06-28:
After updating to 2.201 I get a blank web interface with Safari 10.1.1 and Firefox 53. 

Both browsers show the following in the error console:

    SyntaxError: expected expression, got '<'

     Error: [$injector:modulerr] Failed to instantiate module volumio due to:
             [$injector:nomod] Module 'volumio' is not available! You either misspelled the module
              name or forgot to load it. If registering a module ensure that you specify 
             the dependencies as the second argument.

The initial page loading works, but loading  the linked CSS "styles/app-06a3facd50.css" or the linked script "scripts/app-147f8e29bb.js" returns a "302" redirect to / which most likely triggers the error.

Any help appreciated. Is there any webserver configuration that should be changed or configured?

chsims1, 2017-06-29:
Observation: works fine in Firefox 54

dup2, 2017-06-29:
Just updated to Firefox 54 and still having the same issue.

Are there local any local configuration / overrides via web interface that might have triggered this when updating? Can the assets be regenerated?

volumio, 2017-06-29:
Did you touch any file manually?

dup2, 2017-06-29:
No manual changes on files.

Any way I could re-deploy just the web UI application to ensure I have the correct and latest files for it? Is it /volumio/http/www?

I just tried rebuilding the UI from Volumio2-UI but there I get a lot of deprecation warnings from node.js  and finally a "TypeError: Cannot read property 'prototype' of undefined" when building  in store.js:35 (EventEmitter.prototype)

volumio, 2017-07-04:
You probably have a nodejs version mismatch. We develop it on node 6.1, please use this version

dup2, 2017-07-09:
Rebuilt Volumio2-UI with node 6.11 and redeployed to /volumio/http/www - works fine now.

closed 2017-07-09 by dup2