volumio/Volumio2 #1245: Feature Request: Multi-Room sync'd playback

BLX-S, 2017-06-29:
#953 is also vaguely related to this. I heard Snapcast is a good way to achieve this. It would be cool if you could do a Sonos type thing and pair multiple speakers together and spotify connect to them. You could use the audio sync created by snapcast and add a separate spotify connect entity to it. 
I was already pleasantly surprised volumio.local/ combines all the speakers together in one interface but now I want to simultaneously play music on them (preferably with spotify connect).

chsims1, 2017-06-30:
I believe this is planned as a future feature of Volumio.  For now, you may want to have a look at [https://github.com/Saiyato/volumio-snapcast-plugin](url).

timothyjward, 2019-07-08:
The snapcast plugin is great, but it would be ideal if the integration could be part of the Volumio core. There are also plenty of requests [in the forum](https://forum.volumio.org/multiroom-and-synced-multiroom-should-built-t8264.html) 

Given how cheap it is to set up a Pi Zero W with a DAC/Amp/Speaker it’s realistic to have a home with 2+ instances. What I would like to see is the ability to take a currently playing instance and “send” the audio to one or more other Volumio instances (as visible in the current multi room panel). 

I’m not an expert, but I would expect this to do something like: 
1. Moving the local audio stream for the current instance from the “normal” output device to the snapcast output fifo (starting snapcast server on demand?) 
2. Starting the local snapcast client so that audio continues locally
3. Starting the remote snapcast client and pointing it to the local stream
4. Repeat step 3 for any additional remote targets

I think this would be possible by reusing much of the existing snapcast plugin, and setting up a simple set of sound output devices (local playback, server playback, client playback) as part of Volumio core.

I would be happy to try some things out, but I have limited experience in this area. I’m also happy to try testing at home. Does anyone have any suggestions? @Saiyato ?

volumio, 2019-07-09:
Hi, we are working on that since long time (and hopefully we are almost done with it).
Integrating snapcast is the easy bit: what we are struggling is to send audio from every source to snapcast...

Saiyato, 2019-07-09:
Hi, just my $0.02 as well. 

SnapCast is awesome, don't get me wrong, but it requires resampling which can be expensive and can degrade performance. [Slimstreamer](https://github.com/gimesketvirtadieni/slimstreamer) was the next in line, as it uses the Squeezebox protocol, which doesn't require resampling (and thus HD playback). I've had a setup up-and-running, I even scripted most of it, it's far from mature; but looked promising at the time.

Having said that I've been kept quite busy as of late and most of the time I'm helping out people rather than working on the plugins. I agree, if we're going to move forward with SnapCast, it's needs simpler settings, as the current setup allows for too much configuration which might break (at some point).

As for the rerouting of audio @volumio I used asound.conf to override certain routes, but I agree it's a workaround rather than a solution. ;)

timothyjward, 2019-07-09:
> Hi, we are working on that since long time (and hopefully we are almost done with it).

Great! 😊

Let me know if I can be helpful with testing. I have a bunch of different generation Pis with different DACs and a mixture of wired/wireless networking. 

> SnapCast is awesome, don't get me wrong, but it requires resampling which can be expensive and can degrade performance.

Obviously it’s great if we can avoid any unnecessary audio quality loss, but I think that most users would accept (and understand) a small potential degradation if it only applied when doing multi-room synced streaming. After all, if you’re using WiFi then 32 bit 192kHz audio is probably too much to reliably stream to multiple clients simultaneously. Defaulting to a flat 16bit 48kHz PCM output when doing multi-room would be an issue for some (which hopefully they could configure around), but most users would be unlikely to notice.

> I agree, if we're going to move forward with SnapCast, it's needs simpler settings, as the current setup allows for too much configuration which might break (at some point).

I’m 100% agreed with this. A key value of Volumio is that it “just works” and you don’t have to understand a bunch of esoteric config. The best compliment I can give is that my family can use Volumio without asking for help, or getting annoyed.

badaix, 2020-04-07:
What slimstreamer conceals is that
1. the signal must/will be touched to keep up sync
2. high quality resampling in software usually yields better results than the DACs internal hardware resampling.

Saiyato, 2020-04-14:
Hi @badaix, thanks for clarifying!

So, for sync'd playback the signal must be touched, no matter what method used. The thing that might impact quality is resampling, this is a topic that will probably have as many opinions as it has people reading about it ;)

Key for the current solution is that all inputs will have to be redirected to the fifo-file, maybe some clever ALSA config would allow for such a solution. Because it's less that desirable to (have to keep) patching all players (MPD, Spotify, Shairplay, etc.)

I lack good knowledge of the system to be able to advise at this point, but convergence at ALSA seems the way to go, that way you can have a single point to mute and you only need to capture there.

The players preferably send their progress, but at least metadata on the track to the front-end. It would be easiest if they send to ALSA, which in turn sends to the HW (this poses a problem for a server/client combo though).

Lots to think about and iron out I suppose :)

badaix, 2020-04-14:
Actually, I've never used LMS (I failed to install it back then on a [SheevaPlug](https://en.wikipedia.org/wiki/SheevaPlug)), but the basic idea of not touching the audio signal at all, if possible (i.e. beeing bit perfect) is of course comprehensible. 
Just wanted to say that bit-perfectness per se shouldn't be taken too serious. There are always people noticing the difference between 320kbps MP3s or gold vs copper LAN cables ;)
I also don't know if it's possible to keep different devices in sync when the sample rate changes between tracks (the audio devices must be reopened with the new samplerate)...
To the topic: I was thinking that Volumio is a front end to Mopidy, which is using gstreamer, i.e. that there is only a single audio source (that could even do resampling within the pipeline) that can be redirected to alsa, a file, a pipe and what not.
The alsa sink would be another single point, and it seems to be possible to [redirect alsa (and PulseAudio) to a file](https://github.com/badaix/snapcast/blob/master/doc/player_setup.md#alsa).