volumio/Volumio2 #1259: SMB mount inaccessible

EnerGeh, 2017-07-11:
Hi, first of all: Thanks a lot for this useful piece of software, I love Volumio and so does my wife (it's not easy to impress her with electronic gadgets) ;-)

But I'm still having trouble with SMB mounts and didn't find a solution in the docs:
When mounting a SMB share via WebGUI, the mount gets created in /mnt/NAS/<Alias> and is accessible with root rights in the shell (`ls -l` shows `drwxrws---+ 1 root users 0 Jun 25 19:27 <Alias`). Unfortunately, the mount is inaccessible in "My Music" (unable to update and rescan) and the Music Library where the NAS folder is still empty.
I tried to change the folders permissions `chmod 744 <Alias>` with the following error `chmod: changing permissions of 'OMVMusik/': Read-only file system`. I understood that mounting the share read-only protects it from being modified by Volumio, but in my opinion the intial folder permissions are set incorrectly. Is there a chance to get that fixed in a future version?

If this issue is already addressed, feel free to close this one!

xipmix, 2017-07-11:
You need to establish what are the permissions on the NAS end, ie inside the NAS.

Firstly, at the filesystem level (```ls -ld OMVMusik```). That will govern what volumio ultimately sees.

Next, the export may be 'ro' (readonly) or 'rw' (writes allowed). Look in /etc/exports to see how the exports have been done.

Even if the export is 'rw', the NFS client process will not be able to change the permissions unless the user id number (uid) on the volumio machine (user 'volumio', uid 1000, from memory) is a uid that the exporting machine will allow to make changes, i.e. in your case uid 1000 must be a member of 'users' on the NAS. To check, run ```id``` on the volumio machine.

I think what you want is to run ```chmod a+rx OMVMusik``` on the NAS machine. If you have only web/gui controls, look for something that says 'allow all users' hopefully with an option to make it 'read access for all users'.

Volumio shouldn't be trying to set any restrictive permissions when it does the NFS mount. To find out what has been used, you will have to login to volumio and run ```grep NAS /proc/mounts```.

EnerGeh, 2017-07-11:
Okay, I got that. Just for my information: When I enter correct cifs-credentials in the WebGUI, that's just for accessing the server (and mounting the share), but when I'd like to access files in the mounted folder, the accessing user must have (at least) `rx` rights on the source machine?
I just starting to play with cifs, so please excuse a thousand times my dumb questions :-/

xipmix, 2017-07-11:
> When I enter correct cifs-credentials in the WebGUI,
> that's just for accessing the server (and mounting the share),
> but when I'd like to access files in the mounted folder,
> the accessing user must have (at least) `rx` rights on the source machine?

Correct. To be concrete, if you give cifs creds for 'joe' then
'joe' needs at least rx rights on the source machine.

EnerGeh, 2017-07-16:
@xipmix Thanks a lot for your patience! Got it up and running and learned something about cifs :)

closed 2017-07-16 by EnerGeh