volumio/Volumio2 #1266: Library scan issue

Awazleon, 2017-07-16:
Hello,

I'm using an RPi3, no DAC (it will be buy later), a Syno NAS (NFS mounted) to store my ripped CD AIFF format , all tagged through Python script using Mutagen lib, or manually using iTunes. Cover art are also embedded in the AIFF file.

Using Volumio 2.129, the library scan worked fine. Even if some errors still persist (I guess they come from wrong files in my library) :
Artists : 958, Albums : 1379, Tracks : 13833

Using Volumio 2.201, the scan is incomplete and just a part of the library is displayed with only few albums and few tracks. Around one track per album :
Artists : 131, Albums : 149, Tracks : 157

Right after the scan, I saved a log : 
http://logs.volumio.org/volumio/hjd8yaC.html

I tested and tested again but always got the same result.

So, 
I conducted additional tests with a smaller library.
I created a new share on my NAS with only 3 artists and 10 albums.

Volumio 2.129 scan correctly this new one and found all artists/albums.
I then tested with 2.129 and it found 0 artists, 1 album, 0 track :?:

I opened an SSH session and hereunder is the result :
Jul 12 14:49:16 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:20 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:24 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:28 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:32 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:36 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:40 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:44 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:47 volumio sudo[2745]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Jul 12 14:49:47 volumio sudo[2745]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Jul 12 14:49:48 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:52 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:49:56 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:50:00 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 14:50:03 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: mpd , rescanDb
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003740] ControllerMpd::sendMpdCommand rescan
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003743] sending command...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003749] ControllerMpd::sendMpdCommand status
Jul 12 14:50:03 volumio volumio[939]: info: Mpd Status Update: update
Jul 12 14:50:03 volumio volumio[939]: info:
Jul 12 14:50:03 volumio volumio[939]: [1499871003754] ---------------------------- MPD announces state update
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003761] parsing response...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003765] sending command...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003768] ControllerMpd::getState
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003770] ControllerMpd::sendMpdCommand status
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003772] sending command...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003777] parsing response...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003779] parsing response...
Jul 12 14:50:03 volumio volumio[939]: info: [1499871003785] ControllerMpd::parseState
Jul 12 14:50:04 volumio volumio[939]: info: Command Router : Notfying DB Updatetrue
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004246] ControllerMpd::parseState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004249] ControllerMpd::pushState
Jul 12 14:50:04 volumio volumio[939]: info: CoreCommandRouter::servicePushState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004262] CoreStateMachine::syncState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004269] CorePlayQueue::getTrack 0
Jul 12 14:50:04 volumio volumio[939]: info: STATE SERVICE {"status":"stop","position":null,"seek":null,"duration":null,"samplerate":null,"bitdepth":null,"channels":null,"random":false,"updatedb":true,"repeat":false,"isStreaming":false,"title":null,"artist":null,"album":null,"uri":null}
Jul 12 14:50:04 volumio volumio[939]: info: CURRENT POSITION 0
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004274] CoreStateMachine::syncState stateService stop
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004276] CoreStateMachine::syncState currentStatus stop
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004278] CoreStateMachine::pushState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004280] CoreStateMachine::getState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004281] CorePlayQueue::getTrack 0
Jul 12 14:50:04 volumio volumio[939]: info: CoreCommandRouter::volumioPushState
Jul 12 14:50:04 volumio volumio[939]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004287] interfaceApi::pushState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004289] InterfaceWebUI::pushState
Jul 12 14:50:04 volumio volumio[939]: info: [1499871004320] No code
[... 28 lines cut ...]
Jul 12 14:50:07 volumio volumio[939]: info: [1499871007728] ControllerMpd::parseState
Jul 12 14:50:08 volumio volumio[939]: info: Command Router : Notfying DB Updatefalse
Jul 12 14:50:08 volumio volumio[939]: info: [1499871008124] ControllerMpd::parseState
Jul 12 14:50:08 volumio volumio[939]: info: [1499871008125] ControllerMpd::parseState
Jul 12 14:50:08 volumio volumio[939]: info: [1499871008127] ControllerMpd::pushState
Jul 12 14:50:08 volumio volumio[939]: info: CoreCommandRouter::servicePushState
Jul 12 14:50:08 volumio volumio[939]: info: [1499871008130] CoreStateMachine::syncState
Jul 12 14:50:08 volumio volumio[939]: info: [1499871008132] CorePlayQueue::getTrack 0
Jul 12 14:50:08 volumio volumio[939]: info: STATE SERVICE {"status":"stop","position":null,"seek":null,"duration":null,"samplerate":null,"bitdepth":null,"channels":null,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":null,"artist":null,"album":null,"uri":null}
Jul 12 14:50:08 volumio volumio[939]: info: CURRENT POSITION 0

I can't analyse such log, so if somebody knows... I'll be more than happy to get an explanation and to fix it if this come from my configuration.

Because I don't like to give up without going deeper, I conducted some other tests.
I used the same test library using my other NAS CIFS mounted (An old Netgear guy)
I've got exactly the same result as I've got with the Syno NAS. 2.129 correctly scanned the library and saw all songs while 2.201 not.

I though that could be my song files were "corrupted", so I decided to use Mp3 rather than AIFF but also got the same result whatever the NAS used.

I think I did all tests I can, without reaching a good behaviour from 2.201 :roll:
What is wrong with this version ?
I don't know what I can test more but if one of you have another idea, I'm open.

I would like to use 2.201 because it seems that compilations with Various Artists, are better managed.
Pascal

xipmix, 2017-07-17:
Nice job on poking and prodding things. The log you posted says version 2.233. How did you determine that you are on 2.201 ? The inconsistency might be a bug on its own.

Looking through the log I see 631 instances of 'getMyCollectionStats' which indicates volumio is looking up album data. I'm not sure how many of these appear per track but if it is 4 then that almost matches up with the 157 tracks you report.

I notice that things start to fall over just after you queue an item for playback
```
Jul 12 08:00:24 volumio volumio[943]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 08:00:28 volumio volumio[943]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Jul 12 08:00:30 volumio volumio[943]: info: CoreCommandRouter::volumioGetBrowseSources
Jul 12 08:00:30 volumio volumio[943]: info: CoreCommandRouter::volumioGetQueue
Jul 12 08:00:30 volumio volumio[943]: info: [1499846430781] CoreStateMachine::getQueue
Jul 12 08:00:30 volumio volumio[943]: info: [1499846430783] CorePlayQueue::getQueue
Jul 12 08:00:30 volumio volumio[943]: info:
Jul 12 08:00:30 volumio volumio[943]: [1499846430784] ---------------------------- Client requests Volumio queue
Jul 12 08:00:30 volumio volumio[943]: info: [1499846430786] InterfaceWebUI::pushQueue
Jul 12 08:00:30 volumio volumio[943]: info: [1499846430788] ------------------------------ 9ms
Jul 12 08:00:35 volumio volumio[943]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Jul 12 08:00:35 volumio volumio[943]: info: CURURI: albums://
Jul 12 08:00:35 volumio volumio[943]: Cannot compose Albumart path
Jul 12 08:00:35 volumio volumio[943]: info: No input data
Jul 12 08:00:36 volumio volumio[943]: ERROR getting albumart: Got error: Album not found for Infopath '/data/albumart/web/Felix/%231/info.json'
Jul 12 08:00:37 volumio volumio[943]: ERROR getting albumart: Got error: Album not found for Infopath '/data/albumart/web/Arrested%20Development/3%20Years%2C%205%20Months%20%26%202%20Days%20in%20the%20Life%20of/info.json'
...
```
If you don't do anything, just wait (and wait and wait) until it finishes adding tracks, does it get all the way there? Alternatively, let the count in MyMusic tick over for a few albums, then try to queue something for playback. Does it fail soon after, or keep going for a while?

My guess is there is an issue with getting the albumart embedded in the files; there have been other issues raised about AIFF support (#1246). To test this theory, you could make a small set of files with no embedded art, in a separate NAS folder (it can be a subfolder of your collection, just give the full path to the subfolder when mounting). Unmount all the existing NAS connections and mount only that. Let it scan a few tracks and try queuing an item for playback before it finishes. If it keeps going and completes the library scan, we have a culprit.

Awazleon, 2017-07-17:
Hello,

You're right, I conducted a lot of trial before to update, then after to update, but result remains the same. I'm now back to 2.201 and I can attach a log from it, if this can help.
To sumarize, I started from a fresh SD-Card installation version, then start the scan without doing anything, leaving the scan running during a long time, so I don't understand why there is trace of playback...
Anyway, I also did a test with trial folder and AIFF files without embedded cover art. Result was the same. 2.201 was unable to "detect" the files. It saw only 0 artists, 1 album, 0 track. I made a mistake in the above explaination, 2.129 scanned correctly the files while 2.201 not.
Let me try again and provide new log.
What is the best to attached the log in my comment ? (ok, silly question but I'm not familiar with that...)

xipmix, 2017-07-17:
Best thing is to use the /dev 'send log' button and paste the link here.

Awazleon, 2017-07-18:
Got it : 
http://logs.volumio.org/volumio/icK30Eu.html

xipmix, 2017-07-18:
Thanks for the log. There could well be issues with mpd 0.20.4-1, but I can't see anything in the log that suggests it is the cause. 

I see the NAS getting mounted at 06:30:23 but the first getMyCollectionStats doesn't happen until 06:40:18 which is a little odd - usually when I attach a NAS the library scan starts a few seconds after. I'm guessing that this log comes from this sequence - add NAS, reboot, (possibly manually rescan) and capture the log?

Then you collect the log at 06:41:12 - I guess that is long enough for it to have completed the scan. How many AIFF files are there - there are only 14 instances of getMyCollectionStats.
```
Jul 18 06:41:12 volumio volumio[909]: info: CALLMETHOD: system_controller system sendBugReport [object Object]
```
I can't see any crashes here. I think we need more debug info from getMyCollectionStats.

Awazleon, 2017-07-31:
Ciao,

I tested the new version 2.246 without success. Still have the same issue.
Here is the new log I did : 
http://logs.volumio.org/volumio/c4zZjZK.html

Not sure if this can help. I really don't understand what's happening.
Last point, by using the "Browse Music Library" rather than by Artist or Album, I can access to all my music without problem. So, Volumio can see it but not building a correct Database.
Something that I never said : I use a Micro-SD 4Gb in the RP3. Do I need to use a bigger one ?

Have a nice day

volumio, 2017-07-31:
We are struggling a bit with mpd latest database creation...
If you go to my music there is a new setting called extended file support, try to disable it and see if that helps

Awazleon, 2017-07-31:
I saw it and did a trial but unsuccessful. 
It seems that a fixed version of MPD is coming. To be continued...

volumio, 2017-07-31:
Yep, compiling as we speak

xipmix, 2018-09-05:
HI, is this still an issue for you? Please close this if it's ok now (and let us know the version you tested with)

Awazleon, 2018-09-05:
Sorry, I totally missed it. 
It works fine since a will but I don't remember whiwh version fixed it.

closed 2018-09-05 by Awazleon