volumio/Volumio2 #1268: 2.201 - pushMethod not implemented

GlennV, 2017-07-19:
In the WebSockets API, there is a reference of "pushMethod". 
https://github.com/volumio/Volumio2/wiki/WebSockets-API-Reference#callmethod-on-plugin

I am trying to use "callMethod" to return a config parameter, in this case "volumemax" to handle the UI of my iOS app. I am using the following parameters for this:

```swift
socket.emit(.callMethod, [
  "endpoint": "audio_interface/alsa_controller",
  "method": "getConfigParam",
  "data": "volumemax"
])
```

However, there seems to be no way to get the result back. 

Is this not yet implemented or is there another trick to this?

volumio, 2017-07-23:
callmethod is not meant to give back any data (if not explicitly set on the function you're calling). Your best bet is to use getUiConfig and then parse results

geeks-r-us, 2017-08-14:
Could you please give us a complete example on howto use the pushMethod?
I still struggle with the return value ... even I set it in the called method with defer.resolve(result) 
On the other side the listener for pushMethod is never executed

closed 2017-11-16 by volumio