volumio/Volumio2 #1275: Sleep function doesn't work - Volumio 2.201 on Raspberry pi 3

sparadra1, 2017-07-23:
Hey guys, I tried many time the sleep function but it doesn't work. Nothing happens. In addition, if i choose 9.20 P.M the system will save 9.19 P.M. It always reduces by one minute.
How can i fix this problem ? Thank you.

(I only tried with "Stop music" function)

xipmix, 2017-08-06:
I think you are misunderstanding the function. You are setting the start of a countdown, not a time of day. If you set 'stop in 09:20' and Save, then revisit the Sleep page, the countdown will be at 09:19:44 (say) so the display shows 09:19.

Looking at it now, I can see how it would be easy to get that impression.
This could be avoided by placing the word 'in' between the action field & time selectors.
The popup that appears after setting the timer already says this, but it's easy to miss.

xipmix, 2017-08-06:
If you agree with my comments could you close this issue please - it's really one for Volumio2-UI, I just opened one there.

sparadra1, 2017-08-06:
Ok thank you

closed 2017-08-06 by sparadra1