volumio/Volumio2 #1276: Current solution to use as client is improveable

ShalokShalom, 2017-07-23:
Currently, its a pain to setup the Pi in a way, that i can use it as server and client.
I installed lxde and xorg-server, while it does not make any difference. 

startx: command not found

closed 2017-08-30 by volumio

ShalokShalom, 2017-08-30:
Ignore an issue for one month and close it then: That is why your software is so buggy.

volumio, 2017-08-30:
Closed because it does not make any sense since there is a display plugin which achieves what you want without manually installing lxde

ShalokShalom, 2017-08-30:
Then, you may like to add it: https://volumio.github.io/docs/User_Manual/Quick_Start_Guide.html

volumio, 2017-08-30:
You're right. I will do it now

ShalokShalom, 2017-08-30:
I think also that your super fancy website looks a bit to the contrary to your documentation, which might be the reason why you hide it. Thanks for the nice interaction. +1

volumio, 2017-08-30:
Yep our doc just sucks visually (and also in content)... If you have a suggestion for a better one or can improve a bit the look of it, that would be greatly appreciated

ShalokShalom, 2017-08-30:
Lets think about. Do you love static site generators? Which parts of the documentation are fine?

volumio, 2017-08-30:
Yep I love static generator because they are handy... We just write it and a travis CI task publishes it to gh-pages...
I just love the menu on the left and page on the right, which I guess is common to all docs

ShalokShalom, 2017-08-30:
Gitbook does that. It also offers black themes and so on. For me, the interface of Gitbook sucks.
There is an alternative from Rust, mdbook: https://azerupi.github.io/mdBook/

https://doc.rust-lang.org/book/second-edition/

ShalokShalom, 2017-08-30:
Gitlab hosts nearly each static site, IPFS does too. Github hosts Jekkyl only, afaik.

volumio, 2017-08-30:
Sorry Shalok, nothing really convincing... Maybe a little css tuning on ours?

ShalokShalom, 2017-08-30:
Not really.

ShalokShalom, 2017-08-30:
You want something more fancy?

volumio, 2017-08-30:
If there is,yes. What don't you like of ours?

ShalokShalom, 2017-08-30:
I prefer something without manual CSS handling.