volumio/Volumio2 #1299: Progress ring displays incorrect digital time for very long songs

Turanga1, 2017-08-08:
The progress ring (time circle?) doesn't show the correct digital time for very long songs. According to my testing, "songs" shorter than ~70 minutes work fine. Above ~70 minutes, the digital time display doesn't match the progress ring visual. As seen from the uploaded screenshot, the progress ring is at ~95% of a 76 minute piece and the music played is at ~95% of the piece, but the digital time reads 12:46. Moving the progress ring around will not reproduce the digital time for the same progress ring position, i.e. it could be 12:46 one time and 25:18 the next time.

Confirmed for mp3, m4a (mp4) and flac.

Version 2.246, Pi 2B + Hifiberry DAC+

[image]

xipmix, 2017-08-13:
I think I know why this is occurring. The elapsed time string is computed by this function in ```src/app/services/player.service.js```:
```
  calculateElapsedTimeString() {
    //this.$log.debug(this.elapsedTime);
    let momentDuration = moment.duration(this.elapsedTime),
      hours = momentDuration.hours(),
      minutes = momentDuration.minutes(),
      seconds = momentDuration.seconds();
    if (this.hours > 0) {
      this.elapsedTimeString = hours + ':' + minutes + ':' +
        ((seconds < 10) ? ('0' + seconds) : seconds);
    } else {
      this.elapsedTimeString = minutes + ':' +
        ((seconds < 10) ? ('0' + seconds) : seconds);
    }
  }
```
Notice that ```this.elapsedTimeString``` is coming back in two different forms:
 * for tracks < 60m, 'mm:ss'
 * for tracks longer than that, 'hh:mm:ss'

I suspect whatever displays the elapsedTimeString is showing only the last part, ie 'mm:ss'. Can you confirm that the seconds keep ticking over in your example above?

This might be happening because the font is too large to fit the whole string in the display area or it might be the CSS is not equipped to handle two ```:```  separators.

The display page is ```src/app/themes/volumio/components/track-manager/volumio-track-manager.html``` and the display code:
```
        <div class="bigText">                                                   
          <span ng-if="trackManager.playerService.state.trackType !== 'webradio' && trackManager.      playerService.state.service !== 'analogin'">
            {{trackManager.playerService.elapsedTimeString}}                    
          </span>                                                               
          <div                                                                  
              ng-if="trackManager.playerService.state.trackType === 'webradio'" 
              class="webRadioLabel">                                            
            {{trackManager.playerService.state.trackType}}                      
          </div>                                                                
        </div>                                                                  
```
so I think some more logic is needed to handle these two cases - a smaller font when the elapsed time string is longer than 5 characters.

There's also (arguably) a bug in the calculating function, which this should fix:
```
@@ -140,7 +140,8 @@ class PlayerService {
       minutes = momentDuration.minutes(),
       seconds = momentDuration.seconds();
     if (this.hours > 0) {
-      this.elapsedTimeString = hours + ':' + minutes + ':' +
+      this.elapsedTimeString = hours + ':' +
+        ((minutes < 10) ? ('0' + minutes) : minutes) + ':' +
         ((seconds < 10) ? ('0' + seconds) : seconds);
     } else {
       this.elapsedTimeString = minutes + ':' +
```

xipmix, 2017-08-13:
Alternatively, just ditch the seconds when the elapsed time exceeds 60m
```
@@ -140,8 +140,8 @@ class PlayerService {
       minutes = momentDuration.minutes(),
       seconds = momentDuration.seconds();
     if (this.hours > 0) {
-      this.elapsedTimeString = hours + ':' + minutes + ':' +
-        ((seconds < 10) ? ('0' + seconds) : seconds);
+      this.elapsedTimeString = hours + ':' +
+        ((minutes < 10) ? ('0' + minutes) : minutes);
     } else {
       this.elapsedTimeString = minutes + ':' +
         ((seconds < 10) ? ('0' + seconds) : seconds);
```

xipmix, 2017-08-13:
If the discussion above is correct, this issue more properly belongs to [Volumio2-UI](http://github.com/volumio/Volumio2-UI), but let's wait for more input before considering moving it.

Turanga1, 2017-08-13:
I believe that is correct. I just tested again and I see correct behavior up to 59 min 99 s, then the display starts at 00:00 and continues to count correctly as minutes:seconds without the hour. 

My vote would be to stay consistent with the total time display which is always in minutes. However it would be important to confirm the UI is able to display 3 digit minutes when a pieces is over 100 minutes long. 

I'm not sure why, but once the one hour mark has been crossed by manually moving the progress ring position, the elapsedTimeString is sometimes incorrect, even when the selected time section is less than one hour.

xipmix, 2017-08-13:
> My vote would be to stay consistent with the total time display
> which is always in minutes. However it would be important to
> confirm the UI is able to display 3 digit minutes when a piece
> is over 100 minutes long. 

I agree, I thought of that later.

volumio, 2018-01-28:
Fixed with https://github.com/volumio/Volumio2-UI/pull/376

closed 2018-01-28 by volumio