volumio/Volumio2 #1301: Display year of album when browsing for music?

devhell, 2017-08-09:
Hi,

Just wondering if there is a way to have the year display while browsing collections. Sometimes I'm not always sure which album is the newest of any given artist, so it would be helpful if a year could be displayed in the listings.

Thank you!

liberatorcnx, 2017-08-10:
If you are using your own album art (folder.jpg in each album folder for example) or you are using embedded album art then it would be simple to add the year to the album title.  This will make on-line art lookups fail so don't do this if you are relying on on-line art and add the year AFTER the album title for best sorting.  My albums are listed like:

The Blues Brothers (1976)

I used MP3tag to easily add this to the album titles.

devhell, 2017-08-11:
Thank you @liberatorcnx. I do use online art. I was more thinking of making the stored metadata year available in the music library, so sorting albums by year is an option. So currently when browsing by artist, say I select the band 'A Perfect Circle", then volumio will list the albums of that band, and I have three. It would be helpful if volumio could additionally to the album title also display the year. So instead of:

eMOTIVe
Mer de noms
Thriteenth Step

it lists them as:

Mer de noms (2000)
Thirteenth Step (2003)
eMOTIVe (2004)

Is that possible?

liberatorcnx, 2017-08-21:
@devhell - See my [PR 1311](https://github.com/volumio/Volumio2/pull/1311) which includes this.

devhell, 2017-08-22:
@liberatorcnx, thank you so much! This is excellent! :D :beers:

fabsmusicjunkie, 2018-01-03:
any news on this issue?