volumio/Volumio2 #1303: problem to connect to volumio via web interface

WhistlerAlchemist, 2017-08-11:
Hi,

I have a problem connecting to volumio via the browser (Raspberry Pi 3, Windows 10, FF). I see volumio in my network (list of devices connected to the router), I see the same IP when using ifconfig on the Raspberry, I can ping it from my PC. 

But i cannot open the interface using volumio.local or the IP adress. 

Any suggestions what I could try?

Thanks!

WhistlerAlchemist, 2017-08-11:
hm... might that be because I changed the hostname? If yes, how can I change it back, as the file system is read-only...

xipmix, 2017-08-12:
If you hear the startup sound of volumio, it should be running. Do you hear that after powering up?

If you can connect via ssh you should be able to change the hostname:
```
$ sudo nano /etc/hostname; sudo reboot
```
ie the / filesystem should not be read only. /boot might be, but not the others.

But while you are there, have a look at /var/log/volumio.log to see if there are problems reported there.

Can you say what version of volumio you are running?
```
$ cd /volumio; git log -1
```
might be useful too.

WhistlerAlchemist, 2017-08-12:
Volumio (latest downloadable version) is running (I have an external monitor connected, but the startup sound disabled... ;-) ), and /etc/hostname was indeed read only.

Strange thing was that I could connect to volumio via its hotspot after I turned off my wlan router - but still I could not access the web interface.

Well, I solved the problem by reflashing volumio...

closed 2017-08-30 by volumio