volumio/Volumio2 #1304: Unable to set the default startup volume otherwise the error "Failed to open "alsa" [alsa]; Failed to open ALSA device "softvolume": No such device" occurs

pascal-dev, 2017-08-11:
Rapsberry Model B, Revision 02 with Hifiberry DAC
Volumio version 2.246

This 2.246 version works perfectly except if one tries to define the "default startup volume".

After having set the "Default Startup Volume" to something else than "disabled", it is no longer possible to play a song.
One gets the error : Failed to open "alsa" [alsa]; Failed to open ALSA device "softvolume": No such device.

Nevertheless I found a work around : see below :
1) Go back to the default "Playback Options" :
		Output Device : Audio Jack	
		I2S : off	
	-> Try to play a song : it does not work but one gets no error									
	
2) Set the "output device" to "HifiBerry DAC HiFi pcm5102a-hifi-0"									
	-> Try to play a song : it works but one can not control the volume
	
3) Set "I2S DAC" to "on" and start again the system											
	-> Try to play a song : it works but one can not control the volume

4) Set the "Mixer Type" to "Software"			
	-> Try to play a song: it works and one can control the volume !

(Just think that the volume will be set to zero when starting the system.)			

Thanks for your job :-)

xipmix, 2017-08-12:
Perhaps you can post some debug logs? 
Repeat the steps above, but make a note of the time (nearest minute). 
Then go to http://<volumio>/dev. Press the 'send report' button. (Please avoid characters like ```$!(){}[]``` in the description,  there's a known issue there.) 
You should get a link displayed that points to where the log was uploaded. Paste that here along with the steps and approximate times.

This might also be a known issue in the process of being fixed, please see #1168 #1239 .

pascal-dev, 2017-08-14:
Here are the logs.

14h18 (UTC) :   The "Default Startup Volume" is set to "disabled"
                         Play a song : no problem
14h19 : Set the "Default Startup Volume" to "50"
14h19 : Play a song : KO, error (red message) : "Failed to open "alsa" [alsa]; Failed to open ALSA device "softvolume": No such device"
14h20 : Set the "Default Startup Volume" back to "disabled"
14h21 : Play a song : KO, error (red message) : "Failed to open "alsa" [alsa]; Failed to open ALSA device "softvolume": No such device"
14h22 : Set "I2S DAC" to "off" and set the "output device" to "HifiBerry DAC HiFi pcm5102a-hifi-0"
14h23 : Play a song : OK 
-> Log Part One : http://logs.volumio.org/volumio/tuZbDU5.html

14h24 : Reboot the system
14h34 : Set "I2S DAC" to "on" and set "DAC Model" to "Hifiberry DAC"
14h34 : Play a song : OK
14h37 : Set the "Mixer Type" to "Software" (and "Mixer Control Name" to "Software Volume Control")
14h39 : Play a song : OK
-> Log Part Two : http://logs.volumio.org/volumio/yrRl4Dw.html

luckynrslevin, 2017-08-14:
@pascal-dev I think my bugfix will solve this issue, if you want to test it follow this precedure: https://github.com/volumio/Volumio2/pull/1288#issuecomment-320468485

pascal-dev, 2017-08-15:
@luckynrslevin : I have just tested your bugfix : it works !
Thank you very much !!!

luckynrslevin, 2017-08-16:
@pascal-dev you are welcome, thank you for testing.

marco79cgn, 2017-09-03:
@luckynrslevin 
I applied your fix as well and it works with local music files and web radio stations. Unfortunately it does NOT work with Spotify (spopd) which stays completely silent. Any idea how I can fix this?

**Hardware:** Raspberry Pi Zero, HiFiBerry DAC light Zero

_/etc/spopd.conf_
`[spop]
spotify_username = xxxxxxxxx
spotify_password = yyyyyyyyy
high_bitrate = true
audio_output = sox
pretty_json = true
search_results = 50
cache_path =/run/shm
[sox]
output_type = alsa
output_name = hw:softvolume`

_/etc/asound.conf_
`pcm.softvolume {
    type             plug
    slave.pcm       "softvol"
}
pcm.softvol {
    type            softvol
    slave {
        pcm         "plughw:1,0"
    }
    control {
        name        "SoftMaster"
        card        1
        device      0
    }
max_dB 0.0
min_dB -50.0
resolution 100
}`

_/etc/mpd.conf_
`audio_output {
	type            "alsa"
    name            "alsa"
    device          "softvolume"
    dop             "yes"
    mixer_device    "SoftMaster"
    mixer_control   "SoftMaster"
    mixer_type      "hardware"
}`

_aplay -L_
`null
    Discard all samples (playback) or generate zero samples (capture)
softvolume
softvol
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
default:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dac, 
    Default Audio Device
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dac, 
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Hardware device with all software conversions`

_aplay -l_
`**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0`

volumio, 2017-09-03:
Spotify does not work with softvolume device, use mixer "None"

marco79cgn, 2017-09-03:
Ok, thx. ;(

So Spotify still uses the deprecated libspotify (which might stop working any day), it also doesn't work with softvolume device nor does it play nice with Snapcast (i.e. output to a pipe).

ashthespy, 2017-09-07:
A bit OT, but @volumio  Any particular reason for not supporting softvol with Spotify?

volumio, 2017-09-07:
Only because spop does not want to output to softvol devices... And never found the cause...

ashthespy, 2017-09-07:
@volumio  I have the plugin working on my device (pi zero + PCM5012). Will try and see if the code works fine with my pi3 as well just to confirm I didn't get lucky ;-)

balbuze, 2017-09-07:
There is no problem to use spopd with softvol... But volumio returns a wrong config : output_name = hw:softvolume Must be output_name = softvolume. It is for the same reason I had to use alsaloop in my plugin "volsimpmleequal" because volumio always write "hw" for the output and with virtual device softvol, equal etc) you don't have to use it. It could be a good idea to remove this "hard" code.

volumio, 2017-09-07:
Are you sure that with output_name = softvolume spop actually works?

ashthespy, 2017-09-07:
@volumio [Positive](https://github.com/volumio/volumio-plugins/issues/32#issuecomment-261374978) :-)

marco79cgn, 2017-09-07:
Confirmed! Thx @balbuze 👍 
I also suspected this to be wrong (that's why I wrote "### is this really correct?" above in my spopd.conf). I thought i tried removing the "hw:" before, but somehow i messed up the restart of spopd. After rebooting, it worked with "softvolume".

balbuze, 2017-09-07:
When writing this, I'm using it ;-)

volumio, 2017-09-07:
I did try what @PaulFinch suggested, and edited/etc/spopf.conf there, I initially tired output_name = softvolume but that didn't work. However setting output_name = softvol in /data/plugins/music_service/spop/spop.conf.tmpl fixed my issue :-)

First I need to say that I feel so stupid... Why did I not think about it! Second... what should we put? In the comment above they talk about softvol, but the device is called softvolume

mredone, 2018-01-06:
I have this problem with volumio 2.344.
I use rpi0 with HifiBerry Dac.
I try everything.
VolSpotiConnect not working. Even play flac files form my server.