volumio/Volumio2 #1319: Caracter "&" in names not supported with artist and album image search

Mrpink200, 2017-09-01:
hello

I think i just found a bug
The display album artist and album cover images are not working while matching with last fm when a & symbol is present
for example : 
Chiara Mastroianni & Benjamin Biolay
Iron & Wine and Calexico
Angus & Julia Stone

and same thing with album covers when album name contains a "&" symbol

Thanks for helping
Regards

xipmix, 2017-09-04:
Can you state the problem a little more clearly please? e.g
 * I'm on the 'albums' page of Music Library
 * I search for ```Iron & Wine``` (an album title that is in my collection)
 * the album name is listed in the search results but I see no album art displayed next to the name

Mrpink200, 2017-09-04:
yes, that s it. It happens for every album with a "&" symnol on the album name or artist name

Warter21, 2017-09-04:
there is a same problem with "/"

liberatorcnx, 2017-09-04:
There seems to be a regression in Volumio as it does not include the latest version of album-art.  I pushed a change to album-art that corrected this by replacing '&' with 'and' .  The change was merged 6 months ago...
@volumio - please note!

Meantime, you can correct this behaviour by adding the following line: 

`	var artist = artist.replace ("&", "and");`

to /volumio/node_modules/album-art/index.js.  It should be line 18, immediately before

`	var http = require('http');`

You should backup the old file before editing it.

**Background:** LastFM indexes all artwork with the full word 'and' - it does not recognise artists with '&' (ampersand) in the name.

closed 2017-11-30 by volumio