volumio/Volumio2 #1320: HifiBerry Digi initialization issue v2.246

LifelessDead, 2017-09-03:
Hi all,

Since version v2.2xx there seems to be an initialization issue with (at least) a HifIBerry Digi.

_Typical usage:_ RPi2 + HifiBerry Digi I2S -> Autoplay to resume webradio
Power input is USB port from a Netgear R7000 (which does not power on before router has fully booted)

_Issue symptoms:_
-  Router + RPi2 boots -> music does not start playing
- **No Volumio Start-up sound!**
- 'Playback' UI **unresponsive** when selecting webradio stream / Spotify / local MP3 (song appears in queue, but no song title in playback, buttons don't work, etc)

_Workaround:_
- Power-cycle RPi (boots, autoplays rest of playlist as usual)

_Not reproducible_, happens about 10-20% of the time.
- At first I though it was Autoplay, but the problem is also happens without (thats why it seesms an init issue)
- Then I suspected my HifiBerry, but when switching back to v2.1xx everything worked fine for 2 weeks.

If there is anything I can do to help debugging, let me know...

xipmix, 2017-09-04:
Please send a log from your system
 * power down
 * boot
 * go to http://volumio.local/dev and press the 'send log' button.
   It will prompt for a description, please avoid these characters in that ```!#$&(){}[]``` (known issue, fix is not quite in yet).

If you can do this with a freshly-flashed memory card all the better. but it's not critical. Please let us know whether it was a fresh card or not and which of the symptoms above you observed on the test boot.

LifelessDead, 2017-10-20:
Will have to investigate further... 
Seems like a race condition between router power-up, internet connection available and 'AutoPlay'.

closed 2017-10-20 by LifelessDead