volumio/Volumio2 #1321: about app/i18n/strings_zh.json file

xdsnet, 2017-09-05:
app/i18n/strings_zh.json file have some wrong-words  ,that are not in chinese(zh_cn)

All Good is here:
```
{
  "PLAYBACK_OPTIONS":{
    "PLAYBACK_OPTIONS_TITLE":"播放选项",
    "AUDIO_OUTPUT":"音频输出",
    "OUTPUT_DEVICE_DOC":"输出设备",
    "OUTPUT_DEVICE":"输出设备",
    "ENABLE_I2S_DAC":"启用 I2S DACs",
    "GAPLESS_MP3_PLAYBACK":"无缝播放 MP3",
    "VOLUME_NORMALIZATION":"音量正常化",
    "AUDIO_BUFFER_SIZE":"音频缓冲区大小",
    "BUFFER_BEFORE_PLAY":"播放前缓冲",
    "VOLUME_OPTIONS":"音量选项",
    "DEFAULT_STARTUP_VOLUME":"默认启动音量",
    "MAX_VOLUME_LEVEL":"最大音量值",
    "ONE_CLICK_VOLUME_STEPS":"一键音量步进",
    "VOLUME_CURVE_MODE":"音量曲线模式",
    "NATURAL":"自然",
    "LINEAR":"线性",
    "MIXER_CONTROL":"调音台控制",
    "NO_HARDWARE_MIXER":"没有可用的硬件调音台",
    "MIXER_CONTROLS":"调音台控制",
    "MIXER_CONTROLS_UPDATE":"调音台设定已更新",
    "I2S_DAC_ACTIVATED":"已启用I2S DAC",
    "I2S_DAC_ACTIVATED_MESSAGE":"已经启用, 重新启动生效",
    "I2S_DAC_DEACTIVATED":"禁用I2S DAC",
    "I2S_DAC_DEACTIVATED_MESSAGE":"已经禁用, 重新启动生效",
    "DOP":"DSD 置 PCM (DOP)",
    "DSD_PLAYBACK_MODE":"DSD 播放",
    "DSD_DIRECT":"DSD 直接",
    "MIXER_TYPE":"混音器种类",
    "MIXER_CONTROL_NAME":"混音器名称",
    "AUDIO_RESAMPLING":"音频重采样",
    "TARGET_BITDEPTH":"目标位深度",
    "TARGET_SAMPLERATE":"目标采样率",
    "NATIVE":"Native",
    "SOFTVOL":"软件控制",
    "MIXER_TYPE":"调节类型",
    "SOFTWARE":"软件控制",
    "RESAMPLING_ALG":"重采样段质量",
    "VERY_HIGH":"最高",
    "HIGH":"高",
    "MEDIUM":"中",
    "LOW":"低"
  },
  "NETWORK": {
    "NETWORK_SETTINGS":"网络设定",
    "WIRED_NETWORK":"有线网络设定",
    "AUTOMATIC_IP":"自动获取 IP",
    "STATIC_IP":"静态 IP",
    "NETMASK":"掩码",
    "GATEWAY":"网关",
    "WIRELESS_NETWORK":"无线网络",
    "NETWORK_RESTART_TITLE": "重新启用网络",
    "NETWORK_RESTART_ERROR": "重启网络发生错误: ",
    "NETWORK_RESTART_SUCCESS": "网络重新启用",
    "WIRELESS_RESTART_TITLE": "重新启用无线网络",
    "WIRELESS_RESTART_ERROR": "重启无线网络发生错误: ",
    "WIRELESS_RESTART_SUCCESS": "成功重启无线网络",
    "HOTSPOT_SETTINGS":"热点设置",
    "ENABLE_HOTSPOT":"开启热点",
    "DNS_SETTINGS":"DNS 设定",
    "ENABLE_CUSTOM_DNS":"开启定制DNS服务器"
  },
  "NETWORKFS":{
    "ALIAS_DOC":"选择一个别名显示在NAS的挂载点中"
  },
  "PLAYLIST":{
  "ADDED_TITLE":"已添加",
  "ADDED_TO_PLAYLIST":" 到播放列表 ",
  "ADDED_TO_FAVOURITES":" 到收藏夹 ",
  "REMOVE_ERROR":"不能删除",
  "REMOVE_SUCCESS":"已删除"
  },
  "COMMON":{
    "SAVE":"保存",
    "DISABLED":"禁用",
    "PLAY_TITLE":"播放",
    "PLAY_TEXT":"开始播放 ",
    "ADD_QUEUE_TITLE":"添加到队列",
    "ADD_QUEUE_TEXT_1":"已添加 ",
    "ADD_QUEUE_TEXT_2":" 到队列",
    "REMOVE_QUEUE_TITLE":"删除",
    "REMOVE_QUEUE_TEXT_1":"删除 ",
    "REMOVE_QUEUE_TEXT_2":" 从队列",
    "MY_MUSIC":"我的音乐",
    "CONFIGURATION_UPDATE":"更新配置",
    "CONFIGURATION_UPDATE_DESCRIPTION":"更新配置成功",
    "CONFIGURATION_UPDATE_ERROR":"发生错误",
    "PLAYER_RESTARTED":"播放器重启成功",
    "RESTART":"重启",
    "ON":"开",
    "OFF":"关",
    "SEARCH_ARTIST_SECTION":"艺术家",
    "SEARCH_ALBUM_SECTION":"专辑",
    "SEARCH_SONG_SECTION":"单轨",
    "SCAN_DB_ERROR":"扫描数据库发生错误",
    "SCAN_DB":"扫描音乐",
    "ILLEGAL_CHARACTER_/":"删除禁用的字符 /",
    "FAVOURITES":"收藏夹",
    "PLAYLISTS":"播放列表",
    "MUSIC_LIBRARY":"音乐库",
    "ARTISTS":"艺术家",
    "ARTIST":"艺术家",
    "ALBUMS":"专辑",
    "ALBUM":"专辑",
    "GENRES":"音乐题材",
    "TRACKS":"曲目",
    "CANCEL":"取消",
    "NONE":"无",
    "LAST_100":"last 100",
    "CONTINUE":"继续",
    "SETTINGS_SAVED_SUCCESSFULLY":"保存设置成功",
    "SETTINGS_SAVE_ERROR":"保存设置时发生错误"
  },
  "APPEARANCE":{
    "APPEARANCE":"外观",
    "UI_CONFIG_LANGUAGE":"语言",
    "UI_CONFIG_SELECT_LANGUAGE":"选择语言",
    "NEW_LANGUAGE_SET":"新语言设定",
    "NEW_BACKGROUND_APPLIED":"新的背景已应用",
    "ALBUMART_SETTINGS":"专辑图像设定",
    "WEB_ALBUMART_ENABLE":"网络专辑图像",
    "WEB_ALBUMART_SIZE":"网络专辑图像大小"
  },
  "MAIN_MENU":{
    "MY_MUSIC":"我的音乐",
    "PLAYBACK_OPTIONS":"播放选项",
    "APPEARANCE":"外观",
    "NETWORK":"网络",
    "SYSTEM":"系统",
    "PLUGINS":"插件",
    "ALARM":"闹钟",
    "SLEEP":"睡眠",
    "CREDITS":"致谢",
    "SHUTDOWN":"关机"
  },
  "ALARM":{
    "ALARM_CLOCK_TITLE":"闹钟",
    "SLEEP_MODE_TITLE":"睡眠模式",
    "ALARM_CLOCK_SAVE":"已设置闹钟",
    "SLEEP_MODE_SYSTEM_WILL":"系统将在 " ,
    "SLEEP_MODE_IN":" 在 ",
    "SLEEP_MODE_MINUTE":" 分钟 (s)",
    "SLEEP_MODE_HOUR":"小时(s)",
    "STOP_MUSIC":"停止音乐",
    "TURN_OFF":"关闭"
  },
  "SYSTEM":{
    "SYSTEM_SETTINGS":"系统设置",
    "GENERAL_SETTINGS":"通用设置",
    "PLAYER_NAME":"播放器名称",
    "STARTUP_SOUND":"启动声音",
    "SYSTEM_UPDATES":"系统更新",
    "CHECK_UPDATES":"检测更新",
    "CHECK_UPDATES_DESCR":"检查系统是否有了新版本",
    "DELETE_USER_DATA":"删除用户数据",
    "DELETE_USER_DATA_DESCR":"删除您所有的设置和数据",
    "DELETE_USER_DATA_TITLE":"用户数据删除确认",
    "DELETE_USER_DATA_MESSAGE":"这次删除您所有的个人设置，收藏夹和播放列表 (USB 和 NAS音乐不会被删除). 一旦数据删除，系统会自动重启，您确定继续吗?",
    "FACTORY_RESET":"恢复出厂设置",
    "FACTORY_RESET_DESCR":"把系统恢复到出厂设置吗?",
    "FACTORY_RESET_TITLE":"恢复出厂设置确认",
    "FACTORY_RESET_MESSAGE":"这将把您的设备恢复到出厂设置，您所有的设置会丢失，之后系统会自动重启，您确定要继续?",
    "SYSTEM_CONFIGURATION_UPDATE":"系统配置更新",
    "SYSTEM_CONFIGURATION_UPDATE_SUCCESS":"系统配置成功更新",
    "SYSTEM_NAME":"系统名称",
    "SYSTEM_NAME_ERROR":"无法更改系统名称",
    "SYSTEM_NAME_NOW":"系统名称已更改",
    "UPDATE_ALREADY_LATEST_VERSION":"您的设备已经是最新版本。",
	  "NO_UPDATE_AVAILABLE":"无法升级"
  },
  "WEBRADIO":{
    "MY_WEB_RADIOS":"我的网络电台",
    "FAVOURITE_RADIOS":"电台收藏",
    "TOP_500_RADIOS":"前 500 电台",
    "BY_GENRE_RADIOS":"按照类型",
    "BY_COUNTRY_RADIOS":"按照国家",
    "WEBRADIO":"网络电台"
  }
}
```

volumio, 2017-09-05:
Could you please send us a pull request? And add the missing parts from:
https://github.com/volumio/Volumio2/blob/master/app/i18n/strings_en.json

Thanks!

xdsnet, 2017-09-06:
There is good file:
[strings_zh.zip](https://github.com/volumio/Volumio2/files/1280091/strings_zh.zip)