volumio/Volumio2 #1323: improve softvol quality

balbuze, 2017-09-07:
softvol is often not used because it reduces the sound quality. It is true as it use  a basic linear interpolation to change the volume.
And in general, each time you use a plug in alsa (plughw) a resample is done with the default resampler.
But it can be improve using a other resampler. One of the most interessant is speexrate, better and no to much cpu eater. To use it we need to install libasound2-plugins and then add in asound.conf a line defaults.pcm.rate_converter "speexrate_best". 
source https://github.com/Themaister/alsa-plugins-rsound/blob/master/doc/speexrate.txt

nilsbecker, 2017-09-16:
I would add that this setting could obviate the need for any sampling rate conversion in mpd. In my opinion, alsa is a better place to do resampling than mpd -- for example, input from shairport-sync can be resampled as well, if this is required. there is no significant loss of quality compared to the current way of resampling, since speexrate is of similar quality to soxr.

it seems to me that the best solution is still no resampling and hardware volume control. second best -- resampling by speexrate within alsa; removal of resampling within mpd.

nilsbecker, 2017-09-16:
i just saw that for some reason speex at quality 10 is much slower than soxr at vhq setting in the tests on lastique.github.io . this is surprising to me. all i can say is that if i configure my rpi3 to use speexrate_best, there are no problems doing realtime sample rate conversions from various original rates to 88.2 in stereo.