volumio/Volumio2 #1326: Wifi disconnecting after boot

christemporale, 2017-09-09:
I am running the latest version of Volumio: 2.246. I configured Ethernet and wlan, both with static IPs.
As long as an Ethernet cable is attached, I can access the GUI from both IPs and all works fine. Without Ethernet cable, the system looses the wlan connection. After the boot, the wifi connection is lost after a few seconds (in my router, the raspberry soon disappears from the list of the attached devices).
I had the same issue with version 2.201.
I don't have the issue with RuneAudio. The impossibility of having a working system without cables in the middle of the room is the reason why I use [RuneAudio](http://www.runeaudio.com/).
NOTE: this issue is probably a duplicate of #[668 ](https://github.com/volumio/Volumio2/issues/668) that I cannot reopen (see my comment there).

xipmix, 2017-09-10:
Can you send logs please:
 * power down, disconnect ethernet cable
 * power up, wait for wireless to fail
 * go to /dev page of volumio, press 'send report'

christemporale, 2017-09-10:
Since without any network connection I cannot reach the /dev page, I connected via HDMI, saved journalctl output, made a tarball of /var/log,  and then got the files after a reboot with cable. Please tell me where I can send you those files.

In the meanwhile, at some point I see:
`Sep 10 10:45:47 volumio wpa_supplicant[1299]: wlan0: State: ASSOCIATING -> DISCONNECTED
Sep 10 10:45:47 volumio wpa_supplicant[1299]: nl80211: Set wlan0 operstate 0->0 (DORMANT)
Sep 10 10:45:47 volumio wpa_supplicant[1299]: netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portEnabled=0
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portValid=0
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - EAP success=0
Sep 10 10:45:47 volumio wpa_supplicant[1299]: wlan0: State: DISCONNECTED -> DISCONNECTED
Sep 10 10:45:47 volumio wpa_supplicant[1299]: nl80211: Set wlan0 operstate 0->0 (DORMANT)
Sep 10 10:45:47 volumio wpa_supplicant[1299]: netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portEnabled=0
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portValid=0`

Thank you

xipmix, 2017-09-10:
Ah, silly me. After the wireless fails, if you connect the ethernet again, are you able to reach the /dev page? That would be the quickest way to send the logs. Otherwise, I dunno. pastebin? You should check them & scrub any passwords before pasting them anywhere.

Questions:
 * Do you have the wireless hotspot feature disabled? In your case you probably want it disabled.
 * It would help to know the particular type of wifi device you are using; some have given problems.
Running ```lsusb``` should give enough info. 
 * It may also be helpful to see ```cat /etc/network/interfaces```, with any wifi passwords removed.

Thanks for posting those log lines. I reformatted them below but I am unable to see anything that helps us.
```
Sep 10 10:45:47 volumio wpa_supplicant[1299]: wlan0: State: ASSOCIATING -> DISCONNECTED 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: nl80211: Set wlan0 operstate 0->0 (DORMANT) 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT) 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portEnabled=0 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portValid=0 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - EAP success=0 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: wlan0: State: DISCONNECTED -> DISCONNECTED
Sep 10 10:45:47 volumio wpa_supplicant[1299]: nl80211: Set wlan0 operstate 0->0 (DORMANT) 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT) 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portEnabled=0 
Sep 10 10:45:47 volumio wpa_supplicant[1299]: EAPOL: External notification - portValid=0
```

christemporale, 2017-09-10:
I reconnected the Ethernet cable after a wireless fail (note that the IP wasn't taken even after a "sudo service networking restart" and I had to reassign IP, subnet, GW via ifconfig) and then sent you the logs via dev interface (issue #1326 referenced in the title). 
Here the answers to your questions:
1) Yes, I kept the hotspot disabled
2) 
````
volumio@volumio:~$ lsusb
Bus 001 Device 005: ID 24ae:2010
Bus 001 Device 004: ID 05e3:0736 Genesys Logic, Inc. microSD Reader/Writer
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

````
3) 
```
volumio@volumio:~$ cat /etc/network/interfaces
auto wlan0
auto lo
iface lo inet loopback

allow-hotplug eth0
iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
```

Cheers

xipmix, 2017-09-10:
Great work. Could you please paste the link to the log file, here?
It would have been displayed after you uploaded the logs.

christemporale, 2017-09-11:
[Here ](http://logs.volumio.org/volumio/4hv1Gzy.html)it is (sorry, I wrongly assumed that the send action  would have automatically updated the contributors with the content/link).
Thank you

volumio, 2017-09-11:
Exactly as you describe, the connection is successfully established but its then dropped after few seconds, specifically:
Sep 10 15:06:49 volumio netplugd[967]: wlan0: state ACTIVE flags 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000 -> 0x00001003 UP,BROADCAST,MULTICAST
Sep 10 15:06:49 volumio dhcpcd[953]: wlan0: carrier lost
Sep 10 15:06:49 volumio dhcpcd[953]: wlan0: deleting host route to 192.168.0.20 via 127.0.0.1
Sep 10 15:06:49 volumio dhcpcd[953]: wlan0: deleting route to 192.168.0.0/24
Sep 10 15:06:49 volumio wpa_supplicant[912]: RTM_NEWLINK: ifi_index=3 ifname=wlan0 operstate=2 linkmode=1 ifi_family=0 ifi_flags=0x1003 ([UP])
Sep 10 15:06:49 volumio dhcpcd[953]: wlan0: deleting default route via 192.168.0.1

I noticed you have an hard drive attached to PI, can you please try without the USB Hard drive?

christemporale, 2017-09-11:
I had already tried without the USB drive attached, and the behavior is exactly the same. I hadn't saved the logs, though; I can repeat the experiment if you think it is worth it.
Cheers

volumio, 2017-09-11:
My assumption is that you have a power issue: too much power required by the dongle and the pi can't handle it. Volumio startup is a quite power-intensive task and this can explain the differences among software. In any case, if we implement a reconnect routine that should solve the issue.
No ETA yet for this feature, but as of now you can try one of the following:
- Use a powered USB hub if you have it 
- Power the PI with a more powerful PSU
- Disconnect all USB devices except the dongle 

Let me know

christemporale, 2017-09-12:
I have tried only your third recommendation so far (disconnect all USB devices), but with no luck. 
I will update you with the results of the second one (using a more powerful PSU) once I get such PSU.
Thank you

macmpi, 2017-09-12:
Though I'm unsure what people are reporting here is the exact same issue, there's a known issue with Pis onboard wifi on kernel 4.9.x that is being investigated in length in raspberry kernel discussions (sporadic network losses, particularly on weak signals).

Folks having a wifi dongle can check if connection-loss issue is still here with such dongle:
First, temporarily disable onboard wifi with `dtoverlay=pi3-disable-wifi` in `/boot/config.txt`, and then use your Pi3/PiZero in exact same condition with wifi dongle.
If the issue does not show-up anymore, then it's likely you're hit by the kernel/firmware-related bug (most likely not a Volumio bug).
If the issue still shows-up, then it might be a Volumio issue...

I'm sure once a fixed kernel/firmware is available, it will rapidly make its way into a new Volumio release.

christemporale, 2017-11-05:
I've just flashed your latest Volumio release 2.310 and apparently the issue is fixed: I rebooted several times, and tried to power off and on my PI3 without ethernet cables: now wifi works fine; the system is stable, and also web radios play smoothly.
@macmpi  I see in the changelog that this new version uses a new Linux kernel (4.951), so I think your diagnosis was correct.
Overall, I've also seen many improvements in the user experience on this Volumio version.
Great job guys! Thank you

macmpi, 2017-11-05:
@christemporale glad it helped.
Though the issue I was alluding to is not fully fixed in 4.9.51 kernel yet, it seems there's some progress on that front.

christemporale, 2017-11-18:
After many days I haven't experienced anymore the issue with this Volumio release 2.310 (using the new Linux kernel). 
For me the issue is fixed.
Thank you.

closed 2017-11-18 by christemporale

studio-suman, 2019-06-24:
Hi All,

I am facing this problem with latest version of volumio

xipmix, 2019-08-26:
@studio-suman If you are still seeing this issue and want it addressed can you please
 * create a new issue (and mention this issue - ```#1326``` - in the description so they get linked)
 * try to capture a log and send it in (volumio has a ```volumio logdump``` command that may help)
 * can you confirm you tried the three experiments suggested by volumio, above? What happened?