volumio/Volumio2 #1330: Album divided in several albums because of different artists

biva, 2017-09-14:
I have a compilation album, comprising several artists, for different songs. For example, on the same AlbumA:
- Song1, Artist Bob
- Song2, Artist John

Volumio considers that there are 2 Albums: AlbumA from Bob and AlbumB from John 
For a compilation, this is not what I'd like.

Would it be possible that Volumio considers that as a single album? Or how should I manage tags in order to do that?

biva, 2017-09-14:
My mistake: I should have changed the tags for "Album Artist" and put "Various" for example

closed 2017-09-14 by biva

nilsbecker, 2017-09-16:
I have this problem too and I don't think I did something unusual in the tags.

I have an album in which all songs have the same album title. The artists vary from song to song. There is no tag AlbumArtist set.

Should the unique album name not be enough to display this as one album? Or is it really necessary to go in and manually assign an AlbumArtist entry?

biva, 2017-09-18:
No, you should put an Album Artist. Because an album name could be shared between different albums. This is not a volumio-related issue, this is a general issue.

nilsbecker, 2017-09-18:
well, there is an album artist, as i wrote: 
There is even an album artist tag: Béla Bartók

the spelling of the tag is "album artist". does it need to be capitalized?

biva, 2017-09-18:
No idea, sorry :(