volumio/Volumio2 #1341: SLOW PUSH PLAYLIST TO VOLUMIO

andynz22, 2017-09-20:
I posted this in the Volumio bugs section but not had a response so far.

I have noticed I am getting slow push of playlists. When using Bubbleupnp and selecting play for an album of 10 tracks bubble takes about 6-8 seconds to complete the push. The album then plays normally.

This seems to be related to the bubbleupnp communication with Volumio. Using the same setup/album etc and using other renderers the transfer of playlist happens in 1/2 second. The same when transferring the playlist to OSMC on RP3. Using differnet DLNA servers as source does not effect the time (Twonky & MiniDLNA).

Any ideas on why it is a slow playlist transfer?

RP3 running Volumio 2.246 / Android 6.0 Bubbleupnp licensed 2.8.10 / MiniDLNA/ReadyMedia 1.2.1

Also tried HiFi Cast app and that pushes the playlist without delay so it is a compatability issue between Volumio and current Bubbleupnp.

I did some investigation/sending logs with the Bubble developers and they came back with theses comments:

Volumio uses opensource umpdpcli as renderer:

https://www.lesbonscomptes.com/upmpdcli/

This renderer (volumio) has 2 different ways to be controlled:

- as a regular UPnP/DLNA
- as an OpenHome renderer

The log shows that upmpdcli is controlled as an OpenHome renderer. HiFI cast does not support OpenHome, so it is controlling
it in UPnP/DLNA mode (assuming HiFi cast controls upmpdcli and not smething else running on Volumio).

When you add 40 tracks to upmpdcli in OpenHome mode, there are 40 requests, 1 for each track.
Usually this is fast enough but here the log shows that upmpdcli is very slow to add each track.
There is not much BubbleUPnP can do here. I do not have knowledge of upmpdcli being slow
at inserting tracks but it is possible there is a bug in a version. I do not know what upmpdcli version
is used on Volumio. 

So it looks like an issue with the current version of Volumio and or umpdpcli.

Can the volumio developers comment on this for is.

xipmix, 2017-09-21:
Thanks for reporting. Some logs would help - are you able to send some? Go to http://volumio/dev, press the 'send logs' button and paste the URL you get back into this issue. 

To avoid confounding issues, please could you:
 * power down
 * power up
 * after the volumio 'ready' sound, try to send 10 songs from bubble
 * gather the logs once they are transferred.

andynz22, 2017-09-21:
Hi.  Thanks for replying.  Added 10 tracks twice from BubbleUPNP to volumio, source minDLNA server.
Cheers
Andy
http://logs.volumio.org/volumio/mKxlLVx.html

andynz22, 2017-09-28:
Hi. Have you been able to check the logs I uploaded to see what the problem might be.

Thanks.

xipmix, 2017-09-28:
I don't think I can help, but @volumio might see the problem. I tried the link above and it 404's for some reason...weird.

andynz22, 2017-09-28:
Does @volumio monitor this forum or do they have a direct contact for bugs. I posted in the volumio forums a few weeks back but had no response so not sure how otherwise to contact the developer guys.

volumio, 2017-10-01:
Yes, we do. As stated this might be a bug in upmpdcli itself. Sorry but for us this is low priority, as there isn't much we can do here (and setting openhome to 0 will probably break something else)

andynz22, 2017-10-01:
Thanks for the reply. Is this something I can experiment with, happy to spend some time trying if it helps others too.  Is openhome an integral part of volumio or can it be disabled falling back to upnp mode?  If so can you give me a hint how.

Cheers
Andy

colorcube, 2017-10-26:
Thanks andynz22 for investigating this bug.
There's a bug tracker here: https://opensourceprojects.eu/p/upmpdcli/tickets/
Maybe you put a bug report there too?

andynz22, 2017-10-26:
Hi René

No that wasn't me so must be someone else.

Do you guys have it on the list to look at?

Cheers
Andy