volumio/Volumio2 #1342: The release version(2.246) for RPI has no default locale.

jeffxx, 2017-09-21:
It caused http service unavailable, need default locale for brand new installation.

volumio, 2017-09-22:
I don't get why no default locale should result in http service unavailable, could you clarify why you think so?

jeffxx, 2017-09-22:
I can't connect to volumio with brand new installation. Even I change a different sdcard. 
Also I found someone has encounter the same problem: https://volumio.org/forum/raspberry-not-showing-network-t7431.html

Finally I found the bug by run the command `/usr/local/bin/node /volumio/index.js`.
It will crashed and complained about "can not found locale-undefined.json".
As soon as I copied "local-en.json" to "local-undefined.json", I heard the welcome sound effect. At the same time, volumio ui works fine!

volumio, 2017-09-22:
The problem is that you stopped Volumio on its first boot. 
You should leave it run for 10 minutes on first start

closed 2017-09-22 by volumio

jeffxx, 2017-09-22:
Yes! I had cut it off in 5 minute when I found it can't reached.
But I can't find any description about we must wait 10 minutes on first start. 
Can you add it to the guide? Not only me has encountered this problem.

volumio, 2017-09-22:
https://volumio.github.io/docs/User_Manual/Quick_Start_Guide.html

FIRST BOOT
Volumio's first boot will take usually longer, up to 6 minutes if you're on a Raspberry PI 1. Therefore, be patient of first boot and wait about 5 minutes before starting to use Volumio.
The first time Volumio starts, it will perform some operations:
Extend the filesystem to fill completely your SD Card\ Hard drive
Regenerate SSH keys, to make them unique to your device
Install some packages and configure the system
Generate new thumbnails for the default backgrounds

jeffxx, 2017-09-22:
But not here => https://volumio.org/get-started/