volumio/Volumio2 #1343: Support for JustBoom DAC, Digi and Amp Capes

shawaj, 2017-09-21:
Hi there,

What is the process for getting BeagleBone Capes supported on the BBB version of Volumio? Is it the same sort of thing as for the Raspberry Pi where you just pull in the standard Kernel and then just a few things on top of that to load the drivers correctly and the menu items?

Or is there other stuff that would need doing?

We have got support for our soon-to-be-released JustBoom Capes (DAC, Digi and Amp Capes) in the BeagleBone Kernel:
https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/40698365e569dadf99c40d0b83dd8b3b6dea11b8

Along with the overlays:
https://github.com/beagleboard/bb.org-overlays/commit/3f915e33312c7eaef8b2e0f4f19a187ed0071ce5

And the u-boot stuff:
https://github.com/RobertCNelson/Bootloader-Builder/commit/8363d7f770e89a498d2df1a62c5c734d33e932a5

This should also now be in the latest images, and also should be in apt update / apt upgrade.

For reference, here is the driver patch and the overlays:

[JustBoomCapes.zip](https://github.com/volumio/Volumio2/files/1321932/JustBoomCapes.zip)

Thanks very much in advance for your help!

shawaj, 2017-09-25:
@volumio any ideas?

volumio, 2017-09-25:
Hi @shawaj unfortunately I have no idea how we could support capes for BBB, since the kernel of this device has been developed by a third party contributor (mieero from DIYAUDIO). I guess you could ask him how to integrate. As per Volumio itself, sorry but I think it requires custom code that we don't have interest in doing since BBB is one of the least adopted platform and we don't update it regularly...

shawaj, 2017-09-25:
@volumio do you have the contact details of mieero? Or the GitHub name?

I guess the kernel is just pulled in as standard from the official kernel? 

It works in exactly the same way as RasPi with device tree overlays, so I would imagine the work you have already done for Raspberry Pi support will port over very easily to the BBB.

volumio, 2017-09-25:
@shawaj If it works like pi, then you find a guide on how to add it here:
https://volumio.github.io/docs/I2S_DACs/Adding_Compatibility_to_your_DAC.html

This is his username: https://github.com/miero

shawaj, 2017-09-25:
@volumio ok will sort that file today. The BBB can also do auto
configuration like with the Pi, except the data from the EEPROM would be
read in a slightly different way. How do we handle that?

@miero can you pull in the new kernel?

https://volumio.github.io/docs/I2S_DACs/Adding_
Compatibility_to_your_DAC.html

volumio, 2017-09-25:
@shawaj please note that we will not commit in any work to support the I2S DACs on BBB, we can at most review and accept PR for this platform. I would not bother with auto recognition, but just enable i2s

shawaj, 2017-09-25:
No problem at all.

I'll find out all the information about enabling auto configuration anyway,
and will put it here for future reference even if you don't add it :-)

shawaj, 2017-09-26:
Ok @volumio edited that file - https://github.com/volumio/Volumio2/pull/1346

Info on EEPROM etc coming soon.

@miero just checking you saw this?

shawaj, 2017-10-23:
Some more info on this. For beaglebone overlays are loaded with cape manager, that uses part number to identify overlay. 

Part numbers for JustBoom Capes are:

bb-justboom-amp
bb-justboom-dac
bb-justboom-digi

You can get cape info loaded from eeprom with command:

cat /sys/devices/platform/bone_capemgr/slots

there are cape, name, part number and version, for all content it is nedded to read eeprom image file:

/sys/bus/i2c/devices/2-0057/eeprom

but only part number and version are used to identify overlay.

DAC/Amp I2C address is: 0x4d
Digi I2C address is 0x3b

Mixers are same as our RPi HATS...DAC/Amp have hardware volume control, digi is fixed, only software possible.

@miero does the Volumio BBB system support Cape manager and how would we configure it?

miero, 2017-10-23:
@shawaj published kernel for BBB is not based on the TI 4.4 kernel maintained by RCN, but on the mainline kernel from the kernel.org. So there is no support for cape manager, overlays and other BBB specific stuff.
There was no need for this, because there were no audio capes for BBB and I my personal preference is to be as close with mainline kernel as possible.

Configuration for BBB is done now using DTB file: https://github.com/miero/botic-dev/blob/master-v48/arch/arm/boot/dts/am335x-boneblack-botic.dts

And it was selected at boot in uEnv.txt config: https://github.com/miero/platform-bbb/blob/master/boot/uEnv.txt

The easiest way to integrate your capes (from my side) would be providing separate .dts files that can be put next to the existing.

Switching to RCN kernel would be a problem at this moment, because it does not contain Botic DSD/I2S/SPDIF driver - http://bbb.ieero.com/ and existing users would loose the support.

In a long term view I have to rewrite Botic driver to be more modular and if possible integrated into mainline, but I'm not sure if I manage this in 2017 :-(

Probably the best option at the moment would be to extend platform-bbb of Volumio to contain two kernels (Botic and RCN) and the user would need to edit uEnv.txt to choose the prefered one.

shawaj, 2017-10-23:
@miero I'm guessing you would need the driver patch for our boards as well as the DTS files? I'll send those over shortly for you.

And yes perhaps a good idea to have both kernels as an option but then would also be good to have the Capes supported on both.

Excuse my ignorance but what's the difference between the Botic driver for I2S and the existing one in BBB?

shawaj, 2017-10-24:
@miero Here are driver patch files for all three boards:
[bb-justboom-drivers.zip](https://github.com/volumio/Volumio2/files/1409319/bb-justboom-drivers.zip)

The one called bb-justboom.patch is the full driver files for all boards. The one called bb-justboom-digi.patch patches a few bugs in the first version of the Digi Cape driver so effectively changes some of what is in the bb-justboom.patch file. Hope that makes sense.

Here are the overlays in both .dts source and compiled dtbo
[Overlays-20171024T021119Z-001.zip](https://github.com/volumio/Volumio2/files/1409323/Overlays-20171024T021119Z-001.zip)

Hope that helps?

Thanks

shawaj, 2017-11-16:
@miero just checking you saw this info?