volumio/Volumio2 #1345: AUTO UPDATING MUSIC LIBRARY

nick487, 2017-09-26:
Hi all

I have the latest version of Volumio with a NAS mounted to it. I'm currently using this as a test for some stores this will be used in and we will be regularly be adding new music files to the NAS and need this to update the library automatically, If anyone could help me that would be great.

Thanks

nick-moshulu, 2017-09-28:
Update, I have found the command to use via ssh  'mpc update'. Can someone help me implement that command to run once a day or every few hours. 

Thanks

nick-moshulu, 2017-09-29:
@volumio any ideas?

xipmix, 2017-09-29:
This should do it (not tested)
```
volumio@volumio$ sudo bash
# echo '12 * * * * volumio /usr/bin/mpc update 1>/dev/null 2&1' > /etc/cron.d/mpc-update
# exit
volumio@volumio$ 
```
This will set up a cron job on the volumio to run 'mpc update' hourly, at 12 minutes past the hour.
To make it every few hours, try
```
volumio@volumio$ sudo bash
# echo '12 6,9,12,15,18 * * * volumio /usr/bin/mpc update 1>/dev/null 2&1' > /etc/cron.d/mpc-update
```
This will run at 6:12, 9:12 etc.

It may not work, if it doesn't try sending the output to a log file:
```
# echo '12 12 * * * volumio /usr/bin/mpc update 1>/tmp/mpc.log 2&1' > /etc/cron.d/mpc-update
```
to look for  clues. The '12 12' means run at 12:12  but you can specify whatever time you want.
When you get it working, go back to logging to /dev/null, to save memory.

nick-moshulu, 2017-09-29:
The code ran at 12 past the hour but it still hasn't updated it, is there anything in the code I need to change in order to get it working?

Thanks

xipmix, 2017-10-02:
I'm not aware of anything more but I'm not an mpc expert. The [manual page](https://linux.die.net/man/1/mpc) suggests you might need to specify the directory you want to be (re)scanned.

Did you see anything logged? Note that /tmp gets cleared at reboot, perhaps /var//tmp wuold be a better location for the log file.

nick-moshulu, 2017-10-02:
I don't know if this is the reason it's not working but when I'm looking at the live log, I keep getting this message coming up. 'Unable to open new log files '/var/log/samba/log.smbd' no such file in directory'

nick-moshulu, 2017-10-04:
Added in so it scans the location of where the music is and looks like it worked, I will be doing more testing just to make sure it wasn't a one off. Will keep this updated

nick-moshulu, 2017-10-05:
`Sudo bash
echo '12 * * * * volumio /usr/bin/mpc update > /etc/cron.d/mpc-update
exit`
This works!

nick-moshulu, 2017-10-05:
`sudo bash
echo '12 * * * * volumio /usr/bin/mpc update > /etc/cron.d/mpc-update
exit`

nick-moshulu, 2017-10-05:
`Sudo bash
# echo '12 * * * * volumio /usr/bin/mpc update > /etc/cron.d/mpc-update
# exit`

This works!

nick-moshulu, 2017-10-05:
`Sudo bash`
`echo '12 * * * * volumio /usr/bin/mpc update > /etc/cron.d/mpc-update`
`exit`

This works!

xipmix, 2017-10-06:
Great - if you are satisfied could you please close this issue.

BTW the issue with the samba error messages is well known but not critical, see volumio/Build#93

closed 2017-10-06 by nick-moshulu